Robot-Will/Stino #331: Deviot alternative.

gepd, 2016-01-28:
Hi, As some people may know, Robot-will will release a new version of the plugin but we don't know the date until now. I've been trying to help in the current new-stino branch but as it will be discontinued I'll stop to make changes in the code.

since some time now I've been working in a new plugin for ST called **Deviot**, it's based in the same idea of Stino, but it's using [platformIO](http://platformio.org/) as compiler (and also with other features). The plugin is currently in a beta, if you want to contribute even trying it, you help will be apreciated. The link for the project can be found [here](https://github.com/gepd/Deviot)

MacroMachines, 2016-01-31:
Looks awesome, will be watching it.  will it work with arduino 1.6.7?  Any anticipated timeline for a fully working version?

gepd, 2016-01-31:
It not work with Arduino IDE instead, it does with [platformio](http://platformio.org/) an ecosystem who support more than 200 boards and other features that are coming with the pass of the time.

gepd, 2016-03-13:
@MacroMachines A full working version is already available, you only need to install python in your computer, all the other dependencies are installed automatically (Available officially in the package control)

Robot-Will, 2017-03-11:
@gepd  Thanks for your excellent work and contribition of this software.

closed 2017-03-17 by Robot-Will