Robot-Will/Stino #333: ESP8266 is not supported

DasHammer, 2016-02-02:
With version 1.6.7 of the Arduino IDE you can use the board manager to add support for the ESP8266. It seems that this functionality is not available with stino.

dangsonbk, 2016-02-15:
Same problem, but you can fix this by adding xtensa directory to the environment path:
In my case, add this to PATH

c:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\tools\xtensa-lx106-elf\bin\

Hope this help

tdehaeze, 2016-02-17:
Any idea how to solve this on mac ? :)

DasHammer, 2016-02-20:
@dangsonbk Unfortunatley that didn't work for me. The xtensa directory is different for me: C:\Users\DasHammer\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\bin

Duality4Y, 2016-03-12:
yes but you can still set a path under windows ?

AKBON, 2016-03-17:
I added the Path to the Windows Path environment variable, but it still doesn't work. Any more idea or a more detailed description?

nugrahawahyu, 2016-05-12:
Well i got this error, any idea what is happening?
`[100%] Creating C:\Users\Nugraha\AppData\Local\Temp\Stino_build\Fugue\Fugue.bin...
"C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/esptool/esptool" -eo "C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/bootloaders/eboot/eboot.elf" -bo "C:\Users\Nugraha\AppData\Local\Temp\Stino_build\Fugue/Fugue.bin" -bm dio -bf 40 -bz 4M -bs .text -bp 4096 -ec -eo "C:\Users\Nugraha\AppData\Local\Temp\Stino_build\Fugue/Fugue.elf" -bs .irom0.text -bs .text -bs .data -bs .rodata -bc -ec
Access is denied.
`

gepd, 2016-08-19:
Hey guys I've been working in a plugin called Deviot, currently in version 1.2 wich fully support esp boards, OTA Uptades and a lot of other features. You can know more about the project in: https://github.com/gepd/Deviot

closed 2016-08-27 by gepd