Robot-Will/Stino #335: "avr-ar: : Invalid argument" on Windows 10 with ST3

skyleet, 2016-02-06:
"avr-ar" rcs  "" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\CDC.cpp.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\HardwareSerial.cpp.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\HardwareSerial0.cpp.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\HardwareSerial1.cpp.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\HardwareSerial2.cpp.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\HardwareSerial3.cpp.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\IPAddress.cpp.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\PluggableUSB.cpp.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\Print.cpp.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\Stream.cpp.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\Tone.cpp.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\USBCore.cpp.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\WInterrupts.c.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\WMath.cpp.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\WString.cpp.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\abi.cpp.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\hooks.c.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\main.cpp.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\new.cpp.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\wiring.c.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\wiring_analog.c.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\wiring_digital.c.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\wiring_pulse.c.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\wiring_shift.c.o" "C:\Users\SUNIL YADAV\Documents\Arduino_Build\TestAllPins\wiring_pulse.S.o"
avr-ar: : Invalid argument
[Stino - Error 1]

gepd, 2016-02-06:
which board are you trying to compile?

skyleet, 2016-02-07:
I am using ARDUINO UNO.
Arduino IDE v 1.6.7

gepd, 2016-02-07:
have you installed the `new-stino` branch? I'm using that branch with 1.6.7 and I've not problems compiling with Arduino Uno

skyleet, 2016-02-08:
with the new-stino branch, i can't access menu(verify, upload etc.).
After starting Sublime Text the "Stino-new-stino" folder is moved to "Backup\20160208091531\Stino-new-stino" and menu becomes inaccessible.
Any suggestion please.
[image]

gepd, 2016-02-08:
are you following this instructions to install the plugin?

1. You should first delete whatever version you currently have installed.
2. Open the Command Palette and choose the `Package Control: Add Repository` command.
2. Copy this repo link `https://github.com/Robot-Will/Stino/tree/new-stino` and paste it into the prompt.
3. Open the Command Palette again and run the `Package Control: Install Package` command and choose the package named `Stino`.

skyleet, 2016-02-08:
Actually problem was there was not any .py files in plugin folder(may be moved/deleted by package control.)
So I disabled Package Control.
Menu was working fine after copying StinoSTarter.py and stino folder to plugin directory.
Thanks for helping.
********************************************************
but now another challenge is waiting. Let me try to solve it.
--------------------------------------------------
[Stino - Done building "Blink" in 0.1s.]
[Stino - Start uploading...]
The system cannot find the path specified.

[Stino - Exit with error code 1.]
--------------------------------------------------

skyleet, 2016-02-08:
And the problem is solved after this(thanks to tonyputi for it. https://www.bountysource.com/issues/10627273-don-t-upload-sketch)
-----------------------------------------------------------
add this at line 323 of your stino/pyarduino
target_tools_avrdude_path = os.path.join(ide_path, 'hardware/tools/avr')
and this at 332 of same file
params['runtime.tools.avrdude.path'] = target_tools_avrdude_path
--------------------------------------------------------------------------------------------------
Many many thanks to developers for their hard work and making this awesome plugin. :)

closed 2016-02-08 by skyleet