Robot-Will/Stino #336: Code that compiles in Arduino IDE 1.6.7 breaks in Stino

Marcel-Robitaille, 2016-02-18:
I have code that compiles fine in the Arduino IDE but throws errors in Stino.

Here are my files:

MotorEncoderMaster.ino
````c
#include "Encoders.h"
Encoders * Encoders::instances [2] = { NULL, NULL };

// instances of our class
Encoders right;
Encoders left;

void setup (){
    right.begin (2);   // pin D2
    left.begin (3);   // pin D3
}  // end of setup

void loop (){
    // whatever
}  // end of loop
````

Encoders.h
````c
class Encoders{
    volatile bool switchChanged;

    static Encoders * instances [2];

    static void switchPressedExt0(){
        if(Encoders::instances [0] != NULL){
            Encoders::instances [0]->switchPressed();
        }
    }

    static void switchPressedExt1 (){
        if (Encoders::instances [1] != NULL){
            Encoders::instances [1]->switchPressed();
        }
    }

    public:
        void begin (const int whichPin){
            pinMode (whichPin, INPUT_PULLUP);
            switch (whichPin){
                case 2:
                    attachInterrupt(0, switchPressedExt0, CHANGE);
                    instances [0] = this;
                    break;

                case 3:
                    attachInterrupt(1, switchPressedExt1, CHANGE);
                    instances [1] = this;
                    break;

            }
        }

        void switchPressed(){
            switchChanged = true;
        }

};
````

I get there errors:
````
[Stino - Start building "MotorEncodersMaster"...]
[  3%] Creating C:\Users\marcel\AppData\Local\Temp\Stino_build\MotorEncodersMaster\MotorEncodersMaster.ino.cpp.o...
In file included from C:/Users/marcel/dev/Arduino/MotorEncodersMaster/MotorEncodersMaster.ino:1:0:

C:/Users/marcel/dev/Arduino/MotorEncodersMaster/Encoders.h: In static member function 'static void Encoders::switchPressedExt0()':

C:/Users/marcel/dev/Arduino/MotorEncodersMaster/Encoders.h:7:39: error: 'NULL' was not declared in this scope

         if(Encoders::instances [0] != NULL){

                                       ^

C:/Users/marcel/dev/Arduino/MotorEncodersMaster/Encoders.h: In static member function 'static void Encoders::switchPressedExt1()':

C:/Users/marcel/dev/Arduino/MotorEncodersMaster/Encoders.h:13:40: error: 'NULL' was not declared in this scope

         if (Encoders::instances [1] != NULL){

                                        ^

C:/Users/marcel/dev/Arduino/MotorEncodersMaster/Encoders.h: In member function 'void Encoders::begin(int)':

C:/Users/marcel/dev/Arduino/MotorEncodersMaster/Encoders.h:21:32: error: 'INPUT_PULLUP' was not declared in this scope

             pinMode (whichPin, INPUT_PULLUP);

                                ^

C:/Users/marcel/dev/Arduino/MotorEncodersMaster/Encoders.h:21:44: error: 'pinMode' was not declared in this scope

             pinMode (whichPin, INPUT_PULLUP);

                                            ^

C:/Users/marcel/dev/Arduino/MotorEncodersMaster/Encoders.h:24:59: error: 'CHANGE' was not declared in this scope

                     attachInterrupt(0, switchPressedExt0, CHANGE);

                                                           ^

C:/Users/marcel/dev/Arduino/MotorEncodersMaster/Encoders.h:24:65: error: 'attachInterrupt' was not declared in this scope

                     attachInterrupt(0, switchPressedExt0, CHANGE);

                                                                 ^

[Stino - Exit with error code 1.]
````

Robot-Will, 2017-03-09:
You can follow the dev version Stino2017.

Robot-Will, 2017-03-11:
Just add `#include <Arduino.h>`  at the top of Encoders.h.

```
#include <Arduino.h>

class Encoders{
    volatile bool switchChanged;
    ......
```

closed 2017-03-11 by Robot-Will