Robot-Will/Stino #338: error on compile avr-g++: error: '-DUSB_MANUFACTURER=Unknown': No such file or directory

dajiangqingzhou, 2016-03-05:
HI everybody，thanks for looking my issues!
Sorry for my pool English.

here is the question, when i use the SUBLIME TEXT 3(Build 3103) to compile any arduino file,it return an errot as follow:
C:\Users\XXXX\Documents\Arduino\libraries\IRremote\AnalogReadSerial\AnalogReadSerial.ino.cpp.o...
avr-g++: error: '-DUSB_MANUFACTURER=Unknown': No such file or directory

avr-g++: error: '-DUSB_PRODUCT=Arduino Leonardo': No such file or directory

Thank you all!

arthursdp, 2016-03-07:
Same here using Arduino 1.6.6 and ST3. 

The bug persists after I updated to new-stino, with the latest patch that should have fixed this bug ([d9c538d](https://github.com/Robot-Will/Stino/commit/d9c538d279c6a246fe36cfadf4db56b93e7ba6a8))

gepd, 2016-03-26:
@dajiangqingzhou @arthursdp guys this plugin won't be updated anymore, please read this https://github.com/Robot-Will/Stino/issues/331 with a possible alternative you may like

daconex, 2016-03-29:
This issue us that Stink can't find the Android IDE avr compiler. Fix this by selecting the Arduino IDE application folder. On Windows, it's `C:\Program Files (x86)\Arduino`.
[image]

chkb123456, 2016-04-17:
A possible reason is that in .\preset\platform_avr.txt, some " was replaced with ' . Changing this might help. @dajiangqingzhou 
经验证，把 .\preset\platform_avr.txt 这个文件里的单引号改成双引号，可以解决这个问题，但似乎源代码有误，所以仍无法完成编译，正在研究这个问题。
如果可以的话，使用Uno就没有问题，这个问题只出现在Leonardo等少数板中。

dajiangqingzhou, 2016-04-23:
@chkb123456 thank you！

sibinsivalal, 2016-12-08:
I believe you need to set the **board preference** !
under Menu>Arduino>Board>"_which ever board you are on_" for me its Arduino/Genuino UNO.
[image]

Robot-Will, 2017-03-11:
I have fixed the problems, please try the newest version.

closed 2017-03-11 by Robot-Will