Robot-Will/Stino #340: Issues uploading to an arduino MEGA

jasonlinker, 2016-03-26:
I am new to the Stino package and I am getting the following error

```
[Stino - Start building "Base_Code"...]
[  3%] Creating C:\Users\Jason\AppData\Local\Temp\Stino_build\Base_Code\Base_Code.ino.cpp.o...
The filename, directory name, or volume label syntax is incorrect.

[Stino - Exit with error code 1.]
```

gepd, 2016-03-26:
Hi @jasonlinker please read this: https://github.com/Robot-Will/Stino/issues/331

jasonlinker, 2016-03-27:
So this is a common error/issue with the plugin not my implementation of it?

gepd, 2016-03-27:
Lately this plugin has a lot of erros, most of them because the changes in the Arduino IDE, there was a time when I tried to improve and solve this problems but to me, was better to start with a new plugin not depending on Arduino IDE.

closed 2017-03-09 by Robot-Will