Robot-Will/Stino #341: avr-g++ not found

Thijxx, 2016-04-04:
Got this error. Found the issue here on GitHub here:
https://github.com/Robot-Will/Stino/issues/14

Then read the issue, no fix in there, can't use apt-get since I run Mint and then I get Arduino version 1.0.5
Someone notes that I should move the Arduino directory to my home directory, but that does not make the avr-g++ visible.

Someone else notes something about arduino_avr.txt but I do not have that file on my system.

What is the exact fix for this issue?
Uninstall Arduino 1.6.8 and run 1.0.5? Or can I just make some symlinks somewhere...

Duality4Y, 2016-04-04:
make a symlink to where it expects the compiler to be. that should be a quick fix.

Thijxx, 2016-04-04:
no can do. it kind of concatenates paths so it's a path I can never ever create

Duality4Y, 2016-04-04:
that doesn't make sense? unless i don't understand what you meant, concatenate paths in what way, you meen that symlink does this?

Thijxx, 2016-04-04:
This is what I added to my path:
/home/thijs/arduino-1.6.8/hardware/tools/avr/bin/

But the error is as follows:
bin/sh: 1: /home/thijs/arduino-1.6.8/hardware/tools/avr/bin**//**home/thijs/arduino-1.6.8/hardware/tools/avr/bin/avr-g++: not found

Note the double `/` in the middle between `bin` and `home`.

Duality4Y, 2016-04-04:
how did you add it to your path ?

Thijxx, 2016-04-04:
I added it in `~/.bashrc` and then sourced the file

Duality4Y, 2016-04-04:
what is the command you puth in .bashrc

Thijxx, 2016-04-04:
`export PATH=${PATH}:/home/thijs/arduino-1.6.8/hardware/tools/avr/bin/`

and `echo $PATH` is:
`/home/thijs/.nvm/versions/node/v5.10.0/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/local/go/bin:/opt/mlcp-8.0-4/bin:/home/thijs/arduino-1.6.8/hardware/tools/avr/bin/`

`ls /home/thijs/arduino-1.6.8/hardware/tools/avr/bin/`
avr-addr2line  avr-c++      avrdude      avr-g++        avr-gcc-ar      avr-gcov   avr-ld      avr-nm       avr-ranlib   avr-size     libusb-config
avr-ar         avr-c++filt  avrdude_bin  avr-gcc        avr-gcc-nm      avr-gdb    avr-ld.bfd  avr-objcopy  avr-readelf  avr-strings
avr-as         avr-cpp      avr-elfedit  avr-gcc-4.8.1  avr-gcc-ranlib  avr-gprof  avr-man     avr-objdump  avr-run      avr-strip

Duality4Y, 2016-04-04:
this is verry weird then!

Thijxx, 2016-04-04:
Maybe something in my settings?

```
{
    "arduino_ide_path": "/home/thijs/arduino-1.6.8", 
    "ide.arduino.avr.pro": [
        "ide.arduino.avr.pro.cpu.16MHzatmega328"
    ], 
    "lang_id": "en", 
    "serial_port": "/dev/ttyUSB0", 
    "sketchbook_path": "/home/thijs/Arduino", 
    "target_board_id": "ide.arduino.avr.pro", 
    "target_programmer_id": "ide.arduino.avr.avrisp"
}
```

Duality4Y, 2016-04-05:
these are what my settings look like:
```json
{
    "arduino_ide_path": "/home/robert/Documents/Arduino-Ide/arduino-1.6.7", 
    "bare_gcc": false, 
    "full_compilation": false, 
    "ide.arduino.avr.mega": [
        "ide.arduino.avr.mega.cpu.atmega2560"
    ], 
    "ide.arduino.avr.mini": [
        "ide.arduino.avr.mini.cpu.atmega328"
    ], 
    "ide.arduino.avr.pro": [
        "ide.arduino.avr.pro.cpu.8MHzatmega328"
    ], 
    "ide.esp8266com.esp8266.generic": [
        "ide.esp8266com.esp8266.generic.UploadTool.esptool", 
        "ide.esp8266com.esp8266.generic.CpuFrequency.160", 
        "ide.esp8266com.esp8266.generic.FlashFreq.80", 
        "ide.esp8266com.esp8266.generic.FlashMode.qio", 
        "ide.esp8266com.esp8266.generic.UploadSpeed.115200", 
        "ide.esp8266com.esp8266.generic.FlashSize.4M3M", 
        "ide.esp8266com.esp8266.generic.ResetMethod.ck", 
        "ide.esp8266com.esp8266.generic.Debug.Disabled", 
        "ide.esp8266com.esp8266.generic.DebugLevel.None____"
    ], 
    "lang_id": "en", 
    "serial_port": "/dev/ttyACM0", 
    "sketchbook_path": "/home/robert/Sketchbook/sketchbook", 
    "target_board_id": "ide.arduino.avr.mega", 
    "target_programmer_id": "ide.arduino.avr.avrispmkii", 
    "upload_verbose": true
}
```

NooBJeff, 2016-09-26:
I fixed this by doing the following work:
In file "arduino_compiler.py", comment out the 510 line

the result looks something like that
```
if "avr-" in cmd:
            cmd = cmd.replace('"', '', 2)
            avr = '"%s\\hardware\\tools\\avr' % working_dir
            # cmd = avr + '\\bin\\' + cmd
            cmd = cmd.replace("{runtime.tools.avrdude.path}", avr)
```

closed 2017-03-09 by Robot-Will

barbu110, 2017-03-18:
@NooBJeff where is that file located? I think it's not present anymore since some time passed...

Robot-Will, 2017-03-20:
@barbu110 This file has no longer existed in current version. Thanks.