Robot-Will/Stino #342: Upload problems connected with libraries

Jones-S, 2016-04-12:
I am trying to upload a file via Stino from sublime to my Arduino Yun. First I tried to change my sketchbook to my current github repository, but could not get it to work, because it would not find my libraries associated with the project anymore. (Although I copied the libraries folder into the git repo).
(Maybe you have even a hint for this problem)...

So I changed my sketchbook folder back to the default and tried to upload a sketch from there.
It does find my MQTT library now, but it still throws error with "undefined references".
It opens up my MQTTClient.h file (from the library) and annoys me with errors like:

```
/tmp/Stino_build/lamp_control/core.a(MQTTClient.cpp.o): In function `MQTT::Client<Network, Timer, 128, 0>::readPacket(Timer&)':
/Users/jones/Documents/Arduino/libraries/MQTT/src/lib/MQTTClient.h:398: undefined reference to `MQTTPacket_encode'
/tmp/Stino_build/lamp_control/core.a(MQTTClient.cpp.o): In function `MQTT::Client<Network, Timer, 128, 0>::keepalive()':
/Users/jones/Documents/Arduino/libraries/MQTT/src/lib/MQTTClient.h:619: undefined reference to `MQTTSerialize_pingreq'
/tmp/Stino_build/lamp_control/core.a(MQTTClient.cpp.o): In function `MQTT::Client<Network, Timer, 128, 0>::cycle(Timer&)':
/Users/jones/Documents/Arduino/libraries/MQTT/src/lib/MQTTClient.h:538: undefined reference to `MQTTDeserialize_publish'
/tmp/Stino_build/lamp_control/core.a(MQTTClient.cpp.o): In function `MQTT::Client<Network, Timer, 128, 0>::connect(MQTTPacket_connectData&)':
/Users/jones/Documents/Arduino/libraries/MQTT/src/lib/MQTTClient.h:659: undefined reference to `MQTTSerialize_connect'
/Users/jones/Documents/Arduino/libraries/MQTT/src/lib/MQTTClient.h:671: undefined reference to `MQTTDeserialize_connack'
/tmp/Stino_build/lamp_control/core.a(MQTTClient.cpp.o): In function `MQTT::Client<Network, Timer, 128, 0>::subscribe(char const*, MQTT::QoS, void (*)(MQTT::MessageData&))':
/Users/jones/Documents/Arduino/libraries/MQTT/src/lib/MQTTClient.h:724: undefined reference to `MQTTSerialize_subscribe'
/Users/jones/Documents/Arduino/libraries/MQTT/src/lib/MQTTClient.h:734: undefined reference to `MQTTDeserialize_suback'
collect2: error: ld returned 1 exit status

[Stino - Exit with error code 1.]```

If I am uploading the exact same file from the Arduino IDE, everything works perfectly fine.
Any ideas, what's the problem here?

I would really love not to use the crappy Arduino editor!...

gepd, 2016-08-19:
Hey @Jones-S  I've been working in a plugin called Deviot, currently in version 1.2 I've been working with MQTT for a while and it isn't a problem. You can know more abou the project in: https://github.com/gepd/Deviot

Jones-S, 2016-09-08:
Thanks by the way for your comment. Currently I don't have the time to check it out, but I guess for my next Arduino project I might will. 👍

Robot-Will, 2017-03-09:
You can follow the dev version Stino2017.

closed 2017-03-09 by Robot-Will