Robot-Will/Stino #346: No programmer has been specified on the command line or the config file

PabloVallejo, 2016-05-25:
Hi,

Great plugin, I've installed and specified the arduino port, however when I try to upload the project I get the following error:

```bash
[Stino - Start uploading...]

avrdude: no programmer has been specified on the command line or the config file
         Specify a programmer using the -c option and try again

[Stino - Exit with error code 1.]
```

Thanks on any advice!

Setup:
* Sublime Text 3
* Arduino Uno

eliasu, 2016-06-02:
*PUSH

… same here. Why?

karthikkeyan, 2016-06-05:
Hi,
I dont know the actual solution, i just want to share what i did to upload my sketch successfully, it might help some one to fine the actual solution.

So I tried to upload the my sketch using my sublime text from `Arduino -> Upload` menu. It failed with the same error message as mentioned above. I looked into the sublime console and i got the below command which was trying to upload my sketch file.

```
"/Applications/Arduino.app/Contents/Java/hardware\tools\avr/bin/avrdude" "-C/Applications/Arduino.app/Contents/Java/hardware\tools\avr/etc/avrdude.conf" -v {upload.verify} -patmega328p -carduino "-P/dev/tty.usbmodem1411" -b115200 -D "-Uflash:w:/tmp/Stino_build/blink/blink.hex:i"
```

**Workaround**
I opened my terminal
Navigate to `cd /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin`
Ran the same command with two changes, and it worked.

```
./avrdude "-C/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf" -v  -patmega328p -carduino "-P/dev/tty.usbmodem1411" -b115200 -D "-Uflash:w:/tmp/Stino_build/blink/blink.hex:i"
```

**Changes:**
1. It seems the path to the `avrdude.conf` file have backslash(\) instead of front-slash(/). i changed the path with front-slash(/). 
2. Also i removed `{upload.verify}` from the command. 

These two changes made the command work perfectly.

jshaw, 2016-06-12:
I'm running into the same issue. It worked for me about an hour ago and now I'm getting the message.. I haven't made and drastic changes to my system or anything that I know of.

I'm using Sublime Text 3 and originally the Arduino IDE v1.6.8. With v1.6.8 I was also having the error using the official Arduino IDE + Stino. I upgraded the IDE from v1.6.8 to v1.6.9 and it allowed me to  upload again through the IDE but still I'm running into issues with Stino.

About to try the above solution....

hoonkai, 2016-06-14:
I'm running into the same problem.

dgroo, 2016-06-17:
Same issue here.

dnnssgl, 2016-06-26:
Same here. Tried older versions of Arduino IDE and it seems that 1.6.3 is working fine on MacOS 10.11.5 with Sublime Text 3114.

hnnswldschtz, 2016-06-29:
found at least a workaround! 
open `plattforms.txt` in `Arduino.app/Contents/Java/hardware/arduino/avr/`
in line 106 erase this:  `{upload.verify}`
i can't tell whats the downside so far, but stino and the arduino app itself seems to work fine...
Compared 1.6.7.to 1.6.9., this was added in a newer version.

moe93, 2016-07-23:
The workaround @hnnswldschtz provided works like a charm.

Grisson, 2016-07-24:
The fix @hnnswldschtz provided works like a charm. 
Another tip is that using /dev/tty.*** serial port rather than /dev/cu.*** port. (in mac)

mlohbihler, 2016-07-30:
For me the problem was the version of Arduino AVR Boards. After installing version 1.6.12 i started getting this error. I reverted to 1.6.11 and the error went away. Tools -> Boards -> Boards Manager. Click on Arduino AVR Boards, "select version", and install.

clempat, 2016-08-10:
@mlohbihler I needed to revert to 1.6.10 to make it works !

AbeMont, 2016-08-12:
@clempat  Also reverted back to 1.6.10. It also worked for me.

alexandertmills, 2016-08-27:
I've tried removing "{upload.verify}" and I've tried reverting to the 6 most recent Arduino AVR Board versions. Still no luck. Program works fine in Arduino IDE, still can't upload through Stino. Suggestions?

gepd, 2016-08-27:
please read this: https://github.com/Robot-Will/Stino/issues/331

derjohng, 2016-09-07:
Remove "{upload.verify}" on line 101 of 1.6.11  work for me. 
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/platform.txt

Muzz0, 2016-11-15:
@hnnswldschtz, I think I love you

krishna9830, 2017-02-09:
Go to the arduino menu in sublime, then to board and select your correct board, work for me

Robot-Will, 2017-03-09:
Try follow the dev version Stino2017.

closed 2017-03-21 by Robot-Will