Robot-Will/Stino #354: Cannot use SAMD boards

LeOSW42, 2016-07-18:
Hello world,

I have been using Stino long time before because it was the easyest way to use Arduino on Linux (java IDE is really poor).
Today I try again but I'm using an Arduino Zero board that requires a new version of the Arduino IDE (I'm using Arduino 1.6.9 with the official additional SAMD library).
I cannot access libraries board from Stino menu (only AVR boards).

Stino from https://github.com/Robot-Will/Stino/tree/new-stino (today)
Sublime text 3 (3114)

Thanks

gepd, 2016-08-18:
Hi @LeOSW42 @EduardoPfeifer I've working for a while in a plugin called Deviot, today I've release the version 1.2 with many features. Deviot works with Platformio so isn't necessary to have the Arduino IDE in your computer visit https://github.com/gepd/Deviot to know more about this project.

willbelr, 2016-09-08:
#297

closed 2016-09-19 by gepd