Robot-Will/Stino #360: Add Syntax Highlighting and Coding Convenience

philectron, 2016-08-26:
Hi. I really enjoy using Stino. It's way better than the editor of the Arduino IDE except for the following issues:

**1. Syntax Highlighting:**  
   [image]  
   I would love to have `boolean`, `byte`,  and `word` to be **highlighted** and _italicized_ like other data types shown in the photo.

**2. Coding Convenience:**  
  Let's look at an example, `analogWrite`.  
  As a habit, I type `ana`, then I press Tab, and I type the pin number, then I press Tab. Next, I type the value. The auto-completion makes it easier for me to write my code quickly. Finally, I type `)`, and the editor does a great job when it overwrites the existing `)`. However, when I finish my code with `;`, it doesn't overwrite. And so, I end up like the picture below:  
   [image]  
   I want the `;` to get overwritten like the `)` in auto-complete code.

I'm using **Windows 8.1 Pro**, **Sublime Text 3 build 3114**, and **Stino v2015.12.11.07.03.55**.

I hope you fix some small issue I mentioned above. Stino is a great project. Thanks!

gepd, 2016-08-27:
Hi @philectron, stino hasn't been updated since Aug 3. I started a new project called Deviot, it was based in stino but now it has a lot of new features and I'm always adding new ones and fixing the reported bugs.

If you want to know more about the project visit: https://github.com/gepd/Deviot 

I'll consider your Syntax problem to add in Deviot, I've tested the coding convenience and it seems not affect Deviot

philectron, 2016-08-27:
Thanks a lot! I'll change to Deviot :)

closed 2016-08-27 by philectron