Robot-Will/Stino #361: no programmer has been specified on the command line or the config file

stormbytes, 2016-09-07:
Here's the output log:

[Stino - Start building "blink"...]
Sketch uses 948 bytes (2.9%) of program storage space. Maximum is 32,256 bytes.
Global variables use 9 bytes (0.4%) of dynamic memory, leaving 2,039 bytes for local variables. Maximum is 2,048 bytes.
[Stino - Done building "blink" in 0.0s.]
[Stino - Start uploading...]

avrdude: no programmer has been specified on the command line or the config file
         Specify a programmer using the -c option and try again

[Stino - Exit with error code 1.]

gepd, 2016-09-08:
Hi @stormbytes please read this: https://github.com/Robot-Will/Stino/issues/33

stormbytes, 2016-09-08:
What am I missing?

gepd, 2016-09-08:
I worked for a while in this plugin, the main problem to me was, with each arduino IDE version you should add a new fix to be compatible with the last version. After some time, the autor of stino posted a news about a new plugin he was developing, but none aditional info was released. So I started a new plugin, I took the main code and started to develop a new plugin. As the main goal was do not use the IDE to avoid the same problems with stino, I use PlatformIO which in this moment, has released the version 3 and support more than 270 boards, library manager and other features.

I'm constantly adding new features and fixing bugs. That's just a bit of the story about stino and deviot :)

stormbytes, 2016-09-08:
Wow that looks pretty awesome!

Will definitely try that out.... is that your project?

gepd, 2016-09-08:
The Deviot plugin is my project, PlatformIO isn't

stormbytes, 2016-09-08:
Yes I understand, I've looked at PlatformIO briefly some time back. I was referring to Deviot. So far it looks like it serves the purpose very well !! I will be giving it a run :)

gepd, 2016-09-08:
Thanks @stormbytes if you have any problem please report it to me and I'll try to do my best to solve it :+1:

stormbytes, 2016-09-08:
Definitely! 
I will provide you with some feedback on my impressions and any suggestions. 

Nicely done :)

closed 2016-09-18 by gepd