Robot-Will/Stino #362: Uploading fails: avrdude: invalid baud rate specified '-D'

ryukinix, 2016-09-19:
The sketch compiles ok, but I have a problem about upload. On "Programmer" mode the uploading get timeout and after it exists. Any issue related about that before? Sorry If has, I try search, but was very confuse to me :(

Great project, BTW. I really would be happy if this works in my sublime :D. The oficial IDE is a piece of crap.

gepd, 2016-09-19:
Hi @ryukinix  please read [this](https://github.com/Robot-Will/Stino/issues/331)
Any problem you can open an issue in the Deviot github's an I'll happy to help you

ryukinix, 2016-09-19:
WOW!!! THIS IS AWESOME! Thanks!

ryukinix, 2016-09-19:
(i have some troubles because I use a Arch based distro, but the instructions inside of a issue about that in Deviot solves my problem in some way)

Congratulations! The deviot is really better!

gepd, 2016-09-19:
Can you tell me which issue was that? I hope to release a new version soon so I can prioritize where to spend my time.

Thanks for your support!

ryukinix, 2016-09-19:
Yes, I can. I'll create now a issue on Deviot repository about a new problem whose I'm having troubles right now.

closed 2016-09-23 by gepd