Robot-Will/Stino #363: error on compile: arm-none-eabi-g++

steinerlein, 2016-09-22:
I try to compile for Teensy 3.2 with Teensy 1.29 and Arduino 1.6.8 and get this error:

[Stino - Start building "_160922_logichack_05_teensy_seriell"...]
[  1%] Creating C:\Users\Farid\AppData\Local\Temp\Stino_build\_160922_logichack_05_teensy_seriell\_160922_logichack_05_teensy_seriell.ino.cpp.o...
"arm/bin/arm-none-eabi-g++" -c -O -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=129 -DARDUINO=168 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IE:\01_dreikant\Connected_Office\logichack\_160922_logichack_05_teensy_seriell" "-Ic:\program files (x86)\arduino\hardware\teensy\avr\cores\teensy3" "-Ic:\program files (x86)\arduino\hardware\arduino\avr\variants\standard" "C:\Users\Farid\AppData\Local\Temp\Stino_build\_160922_logichack_05_teensy_seriell\_160922_logichack_05_teensy_seriell.ino.cpp" -o "C:\Users\Farid\AppData\Local\Temp\Stino_build\_160922_logichack_05_teensy_seriell\_160922_logichack_05_teensy_seriell.ino.cpp.o"
Das System kann den angegebenen Pfad nicht finden.
(aka: no such file or directory)

[Stino - Exit with error code 1.]

I have had a look at [https://github.com/Robot-Will/Stino/issues/248](https://github.com/Robot-Will/Stino/issues/248) but that didn't solve it for me.

Any ideas? I would really like to use Stino.

gepd, 2016-09-23:
Hi @steinerlein please read this: https://github.com/Robot-Will/Stino/issues/331

steinerlein, 2016-09-23:
I have successfully resolved all compiler linking problems by hardlinking their respective locations in the arduino IDE's robots.txt and platform.txt. I will post the changes I had to make next week.

KurtE, 2016-10-19:
For the fun of it, I thought it would be nice to try being able to build in Sublime text so I did a search and found Stino...   And I have now run into the issue you mentioned: 

I opened the Example blink, and verified I could compile it for a standard Arduino like a mega.  Then selected Teensy 3.6.  I have Arduino 1.6.12 installed with Teensyduino 1.31-beta1. 

[Stino - Start building "Blink"...]
[  1%] Creating C:\Users\Kurt\AppData\Local\Temp\Stino_build\Blink\core_teensy3\analog.c.o...
"{runtime.hardware.path}/../tools/arm/bin/arm-none-eabi-gcc" -c -O -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD  -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=130 -DARDUINO=1612 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\arduino-1.6.12\examples\01.Basics\Blink" "-IC:\arduino-1.6.12\hardware\teensy\avr\cores\teensy3" "-IC:\arduino-1.6.12\hardware\arduino\avr\variants\standard" "C:\arduino-1.6.12\hardware\teensy\avr\cores\teensy3\analog.c" -o "C:\Users\Kurt\AppData\Local\Temp\Stino_build\Blink\core_teensy3\analog.c.o"
The system cannot find the path specified.

[Stino - Exit with error code 1.]

So it would be great if you shared your solution.   Thanks

KurtE, 2016-10-20:
Update:  I fixed part of this in the python code, to be able to handle the compiler path string as it is in the teensy platform.txt file, plus I added support for extra.time.local which current Teensy builds use. 

There still is an issue with the archive command definition having hard coded path to where the core compile files are stored, which appears to differ from what this build uses.  So I updated my platform.txt to 
change: 
    recipe.ar.pattern="{compiler.path}{build.toolchain}{build.command.ar}" rcs "{build.path}/core/{archive_file}" "{object_file}"

to: 
    archive_file_path={build.path}/{archive_file}
    recipe.ar.pattern="{compiler.path}{build.toolchain}{build.command.ar}"  rcs "{archive_file_path}" "{object_file}"

With these changes I am able to build simple programs like blink. 

However now running into problems with libraries.  In particular, the code does not appear to process the libraries, that were install by Teensyduino.   Which on my machine with the current Arduino the libraries are stored in: C:\arduino-1.6.12\hardware\teensy\avr\libraries

Not sure if another bug report on this or not. 

But in the mean time, I pushed my changes up to: 
https://github.com/KurtE/Stino/tree/new-stino-teensy-support

steinerlein, 2016-10-21:
KurtE, thanks for posting your solution!
I was having a bad conscience for not posting my solution to the problem, but you seem to have come up with a better one by looking at it yourself.
I just hardcoded everything in the teensy\avr\platform.txt that threw an error while compiling. I will not post it all, because it is a very dirty solution to the problem. 

The entries I changed are:
compiler.path=hardcoded location
recipe.ar.pattern: changed "core" to "core_teensy3"
recipe.c.combine.pattern: {build.path}/{archive_file}" became "{build.path}/core_teensy3/{archive_file}"
tools.teensyloader.cmd.path: "{runtime.hardware.path}/../tools" juste became "{compiler.path}"

Again, this is not guaranteed to work with any config, but it did with mine. If you have this problem, Kurts solutions seems much cleaner.

KurtE, 2016-10-22:
Thanks,

Yesterday I verified I could do some uploads with standard teensy loader, then tried setting up Arduino to use TyQt to do the uploads.  Ran into issues where Stino was not picking up the new command that TyQt added: recipe.objcopy.tyqt.pattern

My guess is Arduino enumerates all items under: recipe.objcopy and executes them.  But the current Python code here is hard coded to only look for two specific ones.  I added support to look for the TyQt one as well.   So I was able to get this to run... 

So the only thing I needed to still change in platform.txt is the recipe.ar.pattern, which I showed in previous message. The way I changed it made it still compatible with using Arduino IDE as well. 

Not sure if it will be picked up or not, but did a Pull Request: 
https://github.com/Robot-Will/Stino/pull/365

Robot-Will, 2017-03-09:
@KurtE  What is TyQt?

KurtE, 2017-03-09:
TyQt is now TyCommander is a nice set of utilities for the Teensy processors.  In particular for uploads of programs and Serial monitor.  It is nice for being able to have multiple teensy boards plugged in and it keeping track of which program should go to which teensy and having multiple serial windows open...

More info up at: https://github.com/Koromix/tytools
as well as the forum thread: https://forum.pjrc.com/threads/27825-Teensy-Qt

Robot-Will, 2017-03-09:
@KurtE  Add all recipe.objcopy.*.pattern to build command list, you think it will help?

```
bin_keys = ['recipe.objcopy.eep.pattern']
for key in cmds_info:
     if key.startswith('recipe.objcopy.') and key.endswith('.pattern'):
         if '.eep.' not in key:
              bin_keys.append(key)
for key in bin_keys:
     cmd = cmds_info.get(key, '')
     if cmd:
         cmds.append(cmd)
         msgs.append('')
```

Another question:
In teensy platform.txt, when and how to use the following command?

```
## Patch ELF - TODO: not supported by Arduino 1.6.6 builder
recipe.elfpatch.pattern="{compiler.path}/hardware/tools/{build.elfpatch}" -mmcu={build.mcu} "{build.path}/{build.project_name}.elf" "{sketch_path}/disk"
```

KurtE, 2017-03-10:
Not sure: 

I only learned enough to be dangerous.  But yes I think doing the all objcopy might help.

 

As for patch elf – Not sure.  As mentioned sounds like it is not used by the 1.6.6 builder, which probably implies that it may not 

Be used with the more current versions of the IDE.  I am mostly using 1.8.1  I have also moved to a new machine and don’t fully

Have this stuff properly installed yet.

Robot-Will, 2017-03-11:
@KurtE Thanks a lot.

closed 2017-03-11 by Robot-Will

KurtE, 2017-03-11:
I decided to try and download your latest stuff.  Can I setup to do Teensy?  Looked like you could.

But not sure steps to do so.

 

Thanks

Kurt

 

Closed #363 <https://github.com/Robot-Will/Stino/issues/363> .

Robot-Will, 2017-03-12:
@KurtE As Teensy does not provide a package index file, you need firstly install Arduino IDE and install Teensy to Arduino IDE. Then, set the Arduino IDE folder path as the step in Readme. Then restart SublimeText.

If the Arduino IDE folder path is right (contains hardware folder), you will find 'Arduino IDE' in menu 'platform', and 'teensy' in 'Arduino IDE'. Then choose 'teensy' and choose board, it will work.

As there are still bugs in this soft, I am still work to improve it. Thanks a lot.

KurtE, 2017-03-12:
Thanks, 

I tried again, and had better luck when I moved the Arduino ide install out of the root of the drive:
That is I moved it from: "D:/arduino-1.8.1-T135", 
to: "D:/dev/arduino/arduino-1.8.1-T135", 

I am pretty sure I tried it with / and \ and each time I would reload sublimetext it would blank the field back out. 

I went through the steps as mentioned.  First thing I noticed was the board options were not correct.
That is it wanted to default those to: 
keyboard:  canadian french -> Should be US English
USB: all of the above -> should be just Serial

Tried Open sketch, but the sketch I wanted is in a subfolder.  i.e. 
<documents>/Arduino/Teensy Tests/my sketch

So I used open folder and opened up the sketch folder, I then opened the sketch file and tried a build.
I think it hung, the output showed:
```
Thanks for supporting Stino!
[http://downloads.arduino.cc/libraries/library_index.json] Download started.
[http://downloads.arduino.cc/libraries/library_index.json] 10% (0.22 M / 2.23 M)
[http://downloads.arduino.cc/libraries/library_index.json] 20% (0.45 M / 2.23 M)
[http://downloads.arduino.cc/libraries/library_index.json] 30% (0.67 M / 2.23 M)
[http://downloads.arduino.cc/libraries/library_index.json] 40% (0.89 M / 2.23 M)
[http://downloads.arduino.cc/libraries/library_index.json] 50% (1.12 M / 2.23 M)
[http://downloads.arduino.cc/libraries/library_index.json] 60% (1.34 M / 2.23 M)
[http://downloads.arduino.cc/libraries/library_index.json] 70% (1.57 M / 2.23 M)
[http://downloads.arduino.cc/libraries/library_index.json] 80% (1.79 M / 2.23 M)
[http://downloads.arduino.cc/libraries/library_index.json] 90% (2.01 M / 2.23 M)
[http://downloads.arduino.cc/libraries/library_index.json] Download completed.
[http://downloads.arduino.cc/tools/avrdude-6.3.0-arduino8-i686-w64-mingw32.zip] Waiting for download...
[arduino] avrdude 6.3.0-arduino8: Installation started.
[Build] C:\Users\kurte\OneDrive\Documents\Arduino\Teensy Tests\Kurts_ILI9341_t3n_FB_and_clip_tests...
[Step 1] Check Toolchain.
```
Question: should I continue to make comments on this issue or is there an other preferred way? 

Thanks
Kurt

Robot-Will, 2017-03-13:
1. D:/arduino-1.8.1-T135 is fine, but do not use \ in the path. JSON read this character with error and all settings will be set to defaults, that is blank.

2.  I canceled sort the name of options, and now the defaults will be as your wishes.

3. Sketches I revised, and it will load more directories.

4. I made one copy of Arduino IDE to D: (D:\arduino-1.8.1), set the paths:
{
    "additional_app_path": "d:/arduino-1.8.1",
    "app_dir_path": "{$default}",
    "sketchbook_path": "{$default}"
}

and build the blink.ino:

Thanks for supporting Stino!
[Build] D:\Documents\Arduino\Blink...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[1.8%] Compiling memset.S...
[3.6%] Compiling memcpy-armv7m.S...
[5.5%] Compiling usb_touch.c...
[7.3%] Compiling usb_serial.c...
[9.1%] Compiling usb_seremu.c...
[10.9%] Compiling usb_rawhid.c...
[12.7%] Compiling usb_mtp.c...
[14.5%] Compiling usb_mouse.c...
[16.4%] Compiling usb_midi.c...
[18.2%] Compiling usb_mem.c...
[20.0%] Compiling usb_keyboard.c...
[21.8%] Compiling usb_joystick.c...
[23.6%] Compiling usb_dev.c...
[25.5%] Compiling usb_desc.c...
[27.3%] Compiling touch.c...
[29.1%] Compiling ser_print.c...
[30.9%] Compiling serial6_lpuart.c...
[32.7%] Compiling serial6.c...
[34.5%] Compiling serial5.c...
[36.4%] Compiling serial4.c...
[38.2%] Compiling serial3.c...
[40.0%] Compiling serial2.c...
[41.8%] Compiling serial1.c...
[43.6%] Compiling pins_teensy.c...
[45.5%] Compiling nonstd.c...
[47.3%] Compiling mk20dx128.c...
[49.1%] Compiling math_helper.c...
[50.9%] Compiling keylayouts.c...
[52.7%] Compiling eeprom.c...
[54.5%] Compiling analog.c...
[56.4%] Compiling yield.cpp...
[58.2%] Compiling WString.cpp...
[60.0%] Compiling WMath.cpp...
[61.8%] Compiling usb_inst.cpp...
[63.6%] Compiling usb_flightsim.cpp...
[65.5%] Compiling usb_audio.cpp...
[67.3%] Compiling Tone.cpp...
[69.1%] Compiling Stream.cpp...
[70.9%] Compiling Print.cpp...
[72.7%] Compiling new.cpp...
[74.5%] Compiling main.cpp...
[76.4%] Compiling IPAddress.cpp...
[78.2%] Compiling IntervalTimer.cpp...
[80.0%] Compiling HardwareSerial6.cpp...
[81.8%] Compiling HardwareSerial5.cpp...
[83.6%] Compiling HardwareSerial4.cpp...
[85.5%] Compiling HardwareSerial3.cpp...
[87.3%] Compiling HardwareSerial2.cpp...
[89.1%] Compiling HardwareSerial1.cpp...
[90.9%] Compiling DMAChannel.cpp...
[92.7%] Compiling avr_emulation.cpp...
[94.5%] Compiling AudioStream.cpp...
[96.4%] Compiling Blink.ino.cpp...
[98.2%] Linking everything together...
[100.0%] Creating binary files...
Sketch uses 20,736 bytes (2.0%) of program storage space. Maximum is 1,048,576 bytes.
Global variables use 3,972 bytes (1.5%) of dynamic memory, leaving 258,172 bytes for local variables. Maximum is 262,144 bytes.

5. You may open a new issue for this topic. Thanks.