Robot-Will/Stino #367: Unbind ctrl+alt+1,2,3

AlexisTM, 2016-11-14:
Dear

I have no more access to the #, {, } ), ^ chars because Stino binds `CTRL+Alt+...` keys. 

How should we unbind them?

I tried : 
```
[
  { "keys": ["alt+1", "alt+2", "alt+3", "alt+4", "alt+5", "alt+6", "alt+7", "alt+8", "alt+9", "alt+0"] },
  { "keys": ["ctrl+alt+1", "ctrl+alt+2", "ctrl+alt+3", "ctrl+alt+4", "ctrl+alt+5", "ctrl+alt+6", "ctrl+alt+7", "ctrl+alt+8", "ctrl+alt+9", "ctrl+alt+0"] }
]

``` 

Thanks

AlexisTM, 2016-11-14:
The package Arduino IDE Like on Sublime text is simply out of date and an update make it works again.

Robot-Will, 2017-03-11:
Shortcut is a big problem, So currently I do not set default shortcuts in the new version, but you can set by yourself. Thanks.

closed 2017-03-28 by Robot-Will