Robot-Will/Stino #369: Cannot set Arduino folder on Linux

BondarenkoArtur, 2016-11-23:
When I'm trying to set folder as "/usr/share/arduino"
Ctrl+` gives me
```
Traceback (most recent call last):
  File "/home/uabart/.config/sublime-text-3/Packages/Arduino-like IDE/StinoStarter.py", line 470, in on_done
    app.arduino_info.refresh()
  File "/home/uabart/.config/sublime-text-3/Packages/Arduino-like IDE/app/base.py", line 24, in refresh
    self.genKeywordList()
  File "/home/uabart/.config/sublime-text-3/Packages/Arduino-like IDE/app/base.py", line 53, in genKeywordList
    self.keyword_list += getKeywordListFromPlatform(platform)
  File "/home/uabart/.config/sublime-text-3/Packages/Arduino-like IDE/app/base.py", line 884, in getKeywordListFromPlatform
    keyword_list += getKeywordListFromCoreFolderList(core_folder_list)
  File "/home/uabart/.config/sublime-text-3/Packages/Arduino-like IDE/app/base.py", line 866, in getKeywordListFromCoreFolderList
    cur_keyword_list = getKeywordListFromFile(keywords_file)
  File "/home/uabart/.config/sublime-text-3/Packages/Arduino-like IDE/app/base.py", line 855, in getKeywordListFromFile
    cur_keyword.setType(keyword_type)
UnboundLocalError: local variable 'keyword_type' referenced before assignment
```
I'm using latest version of Manjaro

Robot-Will, 2017-03-09:
You can follow the dev version Stino2017.

Robot-Will, 2017-03-11:
New version do not need to set the Arduino IDE folder.

closed 2017-03-11 by Robot-Will