Robot-Will/Stino #370: ESP8266 boards not found!

ikbelkirasan, 2016-11-24:
Hi, I just tried to install this awesome plugin on my Sublime Text. However, I can't seem to find ESP8266 boards when I try to select them. Is ESP8266 supported?

Thanks

daned33, 2016-11-28:
I also would be interested to know if support for the ESP8266 boards would be added!

Robot-Will, 2017-02-10:
Micropython is more suitable for ESP8266 than Arduino, you can find it at http://www.micropython.org/.

lenghonglin, 2017-02-18:
@Robot-Will   Nope  , i use arduino for esp8266 few day. and i think arduino  is very suitable for esp8266.plz add it to borads. tks

Robot-Will, 2017-02-20:
@lenghonglin "人生苦短,我用Python"  by Bruce Eckel(https://www.zhihu.com/question/19817432, http://sebsauvage.net/python/ )，为什么不试试呢？开发速度快，不用编译，不用一堆工具链。

“Life is short
(You need Python)”

-- Bruce Eckel
ANSI C++ Comitee member,
author of «Thinking in C++» and «Thinking in Java»

BTW: The new version software will add the support of ESP8266.

ikbelkirasan, 2017-02-20:
Python is good but for better performance, C++ becomes way better especially on esp8266.

Robot-Will, 2017-03-09:
(https://github.com/Robot-Will/Stino/blob/Wiki-Images/images/stino_menu01.jpg)

closed 2017-03-09 by Robot-Will

keven425, 2017-04-28:
I'm on Mac. How could I get the Stino version that supports ESP8266?