Robot-Will/Stino #371: Stalled project? State of Stino?

jensens, 2017-01-13:
Symptoms:

- Many open issues
- Some important open pull requests
- Orphaned commit history
- A zoo of forks with herds of branches
- Package Control published software does not work with latest Arduino IDE

What's the state here?

I would like to use the software and willed to help a bit with the plugin, but no idea if its worth nor how.

Robot-Will, 2017-01-17:
Thanks. When I wrote this software, it was arduino 1.0.x. Then, arduino changed to 1.5.x, and changing and changing, all changed. I had no many time to follow the changes, and I waited. So today this software is  obsolete and does not work with latest Arduino IDE. Now arduino 1.8.x seems to be a relative stable version, I am thinking to make a new software based on 1.8.x. I wish it will be done in the following months.

Maxxgold, 2017-01-19:
Please make an updated version : )

juancgarcia, 2017-01-20:
@Robot-Will Would you like any help with the update? Python isn't my best language, but I'm open to lend a hand as well.

wishAI, 2017-01-25:
@Robot-Will Thank you for making all of this. Will you add support for Arduino intel galileo in the new version?

kurnias, 2017-01-31:
@Robot-Will 1.8.x Thank you for crafting Stino, can't wait for Arduino 1.8.x support

Robot-Will, 2017-02-07:
@wishAI I think it will support Arduino intel galileo.

Robot-Will, 2017-02-07:
@juancgarcia Thanks. I will share my thoughts in the wiki, maybe you can do something for this software.

Robot-Will, 2017-02-23:
https://github.com/Robot-Will/Stino/tree/Stino2017, the beginning.

Robot-Will, 2017-03-09:
Now, it works, and I will continue to improve it. You can try it.

closed 2017-03-11 by Robot-Will