Robot-Will/Stino #372: Disabled Menus Verify/Copmpile, Upload

JoinBugs, 2017-01-21:
The first time the Stino package is installed works good but after I closed sublime and gets re-open 
this happens...

[image]

CodyRoberson, 2017-01-27:
If almost everything in the sub-menus is grey'd out then you probably have an invalid settings file (like a bad build path/ app path )

I'm not an expert or very knowledgeable in this, but this issue was quite common for me and still is. Good Luck Friend :)

Just goto  sublime text 3/packages/Arduino-like IDE  and backup -> delete stino.settings

beyond this, it's up to the gods now.

Robot-Will, 2017-03-09:
You can follow the dev version Stino2017.

closed 2017-03-09 by Robot-Will