Robot-Will/Stino #377: Critical error on compile not type "bool"

labastov, 2017-03-09:
I try to compile for ESP8266  and get this error:
[12.9%] Compiling /Users/aa/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/cont.S...
[14.3%] Compiling /Users/aa/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/uart.c...
[15.7%] Compiling /Users/aa/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/time.c...
[17.1%] Compiling /Users/aa/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/libc_replacements.c...
[18.6%] Compiling /Users/aa/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/heap.c...
In file included from /Users/aa/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc_cfg.h:8:0,
                 from /Users/aa/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.h:13,
                 from /Users/aa/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/heap.c:7:
**/Users/aa/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/lib/gcc/xtensa-lx106-elf/4.8.2/plugin/include/debug.h:63:3: error: expected specifier-qualifier-list before 'bool'
   bool (* ignore_block) (const_tree);**

Arduino IDE compile without error.
Any ideas?

Robot-Will, 2017-03-10:
No, and I will find it out.

Robot-Will, 2017-03-10:
Same problem on LinuxMint. Type name 'bool' is not defined.
On Win10 no this problem.

```
In file included from /home/robot/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc_cfg.h:8:0,
                 from /home/robot/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.h:13,
                 from /home/robot/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/heap.c:7:
/home/robot/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/lib/gcc/xtensa-lx106-elf/4.8.2/plugin/include/debug.h:63:3: error: expected specifier-qualifier-list before 'bool'
   bool (* ignore_block) (const_tree);
   ^
/home/robot/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/lib/gcc/xtensa-lx106-elf/4.8.2/plugin/include/debug.h:155:55: error: unknown type name 'bool'
                                                  int, bool);
                                                       ^
/home/robot/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/lib/gcc/xtensa-lx106-elf/4.8.2/plugin/include/debug.h:160:43: error: expected ')' before 'int'
 extern void debug_nothing_tree_int (tree, int);
                                           ^
/home/robot/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/lib/gcc/xtensa-lx106-elf/4.8.2/plugin/include/debug.h:162:1: error: unknown type name 'bool'
 extern bool debug_true_const_tree (const_tree);
 ^
/home/robot/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/lib/gcc/xtensa-lx106-elf/4.8.2/plugin/include/debug.h:183:1: error: unknown type name 'bool'
 extern bool dwarf2out_do_frame (void);
 ^
/home/robot/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/lib/gcc/xtensa-lx106-elf/4.8.2/plugin/include/debug.h:184:1: error: unknown type name 'bool'
 extern bool dwarf2out_do_cfi_asm (void);
 ^
```

labastov, 2017-03-10:
I have MacOS.
But Arduino IDE compile without error.

Must be we are need  declare anything path to libraries.
How I can diagnosting this problem?

I see different compiling key for Stino and Arduino IDE:
  
# Stino:
"/Users/lva/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" \
-D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ \
"-I/Users/lva/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" \
"-I/Users/lva/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include" \
"-I/Users/lva/Library/Arduino15/build/Blink/core" \
-c 
###  -w 
-Os -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -std=gnu99 -ffunction-sections -fdata-sections \
-DF_CPU=160000000L -DLWIP_OPEN_SRC   \
### -DARDUINO=20000 \
-DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_esp8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266  \
"-I/Users/lva/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/include" \
"-I/Users/lva/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/lib/gcc/xtensa-lx106-elf/4.8.2/include" \
"-I/Users/lva/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/lib/gcc/xtensa-lx106-elf/4.8.2/install-tools/include" \
"-I/Users/lva/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/lib/gcc/xtensa-lx106-elf/4.8.2/plugin/include" \
"-I/Users/lva/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include" \
"-I/Users/lva/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/sysroot/usr/include" \
"-I/Users/lva/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/esp8266/examples/Blink" \
"-I/Users/lva/Library/Arduino15/build/Blink" \
"-I/Users/lva/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266" \
"-I/Users/lva/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/spiffs" \
"-I/Users/lva/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc" \
"-I/Users/lva/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/libb64" \
"-I/Users/lva/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/variants/generic" 
"/Users/lva/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/heap.c" 
-o "/Users/lva/Library/Arduino15/build/Blink/heap.c.o"

# ARDUINO IDE

"/Users/lva/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" 
-D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ 
"-I/Users/lva/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" 
"-I/Users/lva/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include" 
"-I/var/folders/6f/rqc8dsq11vl_1k9kptr8cl3r0000gn/T/arduino_build_638004/core" 
-c 
###-Wall -Wextra 
-Os -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -std=gnu99 -ffunction-sections -fdata-sections 
-DF_CPU=160000000L -DLWIP_OPEN_SRC   
###-DARDUINO=10801 
-DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 
"-I/Users/lva/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266" 
"-I/Users/lva/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/variants/generic" 
"/Users/lva/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/heap.c" 
-o "/var/folders/6f/rqc8dsq11vl_1k9kptr8cl3r0000gn/T/arduino_build_638004/core/heap.c.o"

How i can correct its?

Robot-Will, 2017-03-10:
I see your are using MAC, but I have not.

The compilation paramers are from the platform.txt file(/Users/lva/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/platform.txt), so you can find -w from this file. -w and -Wall -Wextra are gcc compilation parameters.

```
-w Inhibit all warning messages. 

-Wall turns on the following warning flags:
-Wextra
This enables some extra warning flags that are not enabled by -Wall. (This option used to be called -W. The older name is still supported, but the newer name is more descriptive.)
```

10801 is the version of Arduino IDE, stands by 1.8.1. So I give a higher number.

I think the above differences are not the key to solve the problem, I will to check the Arduino Builder source code to find out the differences if I have enough time. Thanks.

On windows, there is no such the file (plugin/include/debug.h), so it works.

labastov, 2017-03-10:
What is 2.0.0 Arduino IDE version? I download latest night version is 1.8.2.
Do you meed  EPS8266  platform 2.0.0 version?

Robot-Will, 2017-03-10:
```
## Compile c files
recipe.c.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.cpreprocessor.flags} {compiler.c.flags} -DF_CPU={build.f_cpu} {build.lwip_flags} {build.debug_port} {build.debug_level} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} -DARDUINO_BOARD="{build.board}" {compiler.c.extra_flags} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"
```

In platform.txt, -DARDUINO={runtime.ide.version} needs a number. This software does not depend on Arduino IDE, so it does know the Arduino IDE's version. 

In early Arduino IDE version before 1.0, the header file is not Arduino.h, and this header is imported by IDE, so for compatibility of old codes, the build process adds this version number. So I gave a larger number.

Robot-Will, 2017-03-21:
I fixed this problem. Thanks a lot.

closed 2017-03-21 by Robot-Will