Robot-Will/Stino #378: Build for Teensy 3.6 hangs Stino2017

KurtE, 2017-03-13:
Looks like a lot of great progress on improving this! 

As was mentioned in the issues #363 I made the changes you mentioned in the last posting in that thread.  I set up the directories to point to my Arduino install, plus where the app_dir_path is.  I left the sketchbook path as default.

I reloaded sublime text 3, configured to compile for Teensy 3.6. 
I was able to load up the example blink.ino and was able to successfully build it 👍 

I then tried a more complex program, which hangs:
```
[Build] C:\Users\kurte\OneDrive\Documents\Arduino\Teensy Tests\Kurts_ILI9341_t3n_FB_and_clip_tests...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
```
The program is a modified version of the example program that is up in my github project:
https://github.com/KurtE/ILI9341_t3n

I have that project installed on my machine at my Sketch folder \ libraries\ili9341_t3n.
The example program includes that library as well as 

It also relies on my SPIN library which is also a project up on my github and it also is installed in my Sketch library folder. 

And likewise it relies on the spi library.  Which the arduino ide finds two version (One in my library folder and one in d:\arduino-1.8.1\hardware\teensy\avr\libraries\spi, which is not used. 

In particular the .ino file includes: 
```
#include <ili9341_t3n_font_Arial.h>
#include <ili9341_t3n_font_ArialBold.h>
#include <ILI9341_t3n.h>

#include <SPIN.h>
#include "SPI.h"
```
The three ili9341... files are all in the ili9341_t3n library. 

Let me know if there is something I should try out and/or you need additional information.

Kurt

Robot-Will, 2017-03-13:
I fixed this problem.
But the example graphicstest did not pass the build. The error message is:

```
graphicstest" -larm_cortexM4lf_math -lm
d:/arduino-1.8.1/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/lib/armv7e-m/fpu\libc.a(lib_a-sbrkr.o): In function `_sbrk_r':
sbrkr.c:(.text._sbrk_r+0xc): undefined reference to `_sbrk'
collect2.exe: error: ld returned 1 exit status
```

I found some links for this problem:
http://stackoverflow.com/questions/13235748/linker-error-on-a-c-project-using-eclipse
https://sites.google.com/site/stm32discovery/open-source-development-with-the-stm32-discovery/getting-newlib-to-work-with-stm32-and-code-sourcery-lite-eabi

KurtE, 2017-03-13:
Thanks, your fix appears to have fixed my build…  Not sure about the _sbrk… 

 

Will try loading the graphics test and see if I get it as well and see if I  can figure out why.

Note: I am using the beta release of Teensyduino with the newer compiler. 

 

Thanks again

 

http://stackoverflow.com/questions/13235748/linker-error-on-a-c-project-using-eclipse
https://sites.google.com/site/stm32discovery/open-source-development-with-the-stm32-discovery/getting-newlib-to-work-with-stm32-and-code-sourcery-lite-eabi

KurtE, 2017-03-13:
I was able to reproduce this one.  Again not sure why in this case this happens but not in my other program?

 

What I have noticed is that there are differences with your build process than with the Arduino and I am not sure if that is an issue or not. 

That is with the Arduino build, only the Core files are gcc-ar into core.a file 

 

That is for example the SPI library – With the Arduino build, it looks like the same gcc like command is issued to compile, but on Arduino it is not gcc-ar to the .o file, but instead passed directly onto the link command. 

 

Also when I am doing a build in Stino, I am getting a few more comments like:

```

[100.0%] Creating binary files...

"D:/arduino-1.8.1/hardware/teensy/../tools/arm/bin/arm-none-eabi-gcc" -O2 -flto -fno-fat-lto-objects -Wl,--gc-sections,--relax,--defsym=__rtc_localtime=1489419182 "-TD:/arduino-1.8.1/hardware/teensy/avr/cores/teensy3/mk66fx1m0.ld" -fuse-linker-plugin -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -o "D:/Users/kurte/AppData/Local/Arduino15/build/graphicstest/graphicstest.elf" "D:/Users/kurte/AppData/Local/Arduino15/build/graphicstest/ili9341_t3n_font_ArialBold.c.o" "D:/Users/kurte/AppData/Local/Arduino15/build/graphicstest/core.a" "-LD:/Users/kurte/AppData/Local/Arduino15/build/graphicstest" -larm_cortexM4lf_math -lm

d:/arduino-1.8.1/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: mk20dx128.c.o: plugin needed to handle lto object

d:/arduino-1.8.1/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: Print.cpp.o: plugin needed to handle lto object

d:/arduino-1.8.1/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/lib/armv7e-m/fpu\libc.a(lib_a-sbrkr.o): In function `_sbrk_r':

sbrkr.c:(.text._sbrk_r+0xc): undefined reference to `_sbrk'

collect2.exe: error: ld returned 1 exit status

```

When LTO compile is turned on…   Will continue to play 

 

http://stackoverflow.com/questions/13235748/linker-error-on-a-c-project-using-eclipse
https://sites.google.com/site/stm32discovery/open-source-development-with-the-stm32-discovery/getting-newlib-to-work-with-stm32-and-code-sourcery-lite-eabi

KurtE, 2017-03-14:
FYI - Still trying to figure out why compiling my own test program works OK, which uses most of the same libraries as graphictest... Again probably just an order issue in the link. 

Some other quick questions/comments

a) I am assuming that this is still a WIP and the code is not in place to mark errors in source files is not there yet. 

b) Wish the code that runs every 30 minutes to download list of boards... was configurable and had option to only do at startup.  Having the window suddenly change can be distracting. 

c) Had an issue yesterday, where I load up Sublimetext3, then do an Open Folder command which opens a new window.  If I then do a build from this new window it does all of it's output on the window that opened when I first started the program.  And if I close that window, then the messages did not show up at all... 

Now back to trying things out.   And thanks for doing all of this!

Robot-Will, 2017-03-15:
Thanks for your suggestion, still there are many works to do, I am working on it.

Robot-Will, 2017-03-17:
Finally, I found an error for searching cpp files, defaultly it should not search with recursion, but it does it. So cpp files should which should not be compiled are added to build list. Now I fixed it. Thanks a lot.

closed 2017-03-17 by Robot-Will