Robot-Will/Stino #381: Fails on STM32L4 upload using DFU programmer

Defragster, 2017-03-16:
Started using Sublime after note on Teensy Forum - installed Stino-Stino2017 and it worked on the Teensy 3.6 sketch I uploaded using the external program app! - Thks

For STM32 it found my JSON and allowed a full build from : https://grumpyoldpizza.github.io/arduino-STM32L4/package_STM32L4_boards_index.json

The build worked but it did not show signs of attempting or failing the DFU Upload.

Using the Ladybug variant. Here is the : https://github.com/GrumpyOldPizza/arduino-STM32L4

Defragster, 2017-03-17:
Like on Teensy it uses a stand alone uploader for the STM DFU interface.  If I can provide any other details let me know.

Robot-Will, 2017-03-17:
I have no STM32 boards, and I have no idea how to upload sketch to it. This soft depends on platform.txt, the upload command is from the file. So the upload command should be right.

I found two commands in platform.txt, and now I added this to commands after objcopy. Perhaps it will help. Thanks.

```
## Append DFU suffix (dfu file)
recipe.hooks.objcopy.postobjcopy.1.pattern="{compiler.platform.path}/{compiler.suffix.cmd}"  -v "{build.vid}" -p "{build.pid}" -d "{build.did}" -a "{build.path}/{build.project_name}.dfu"

## Append DFU suffix (iap file)
recipe.hooks.objcopy.postobjcopy.2.pattern="{compiler.platform.path}/{compiler.suffix.cmd}"  -v "{build.vid}" -p "{build.pid}" -d "{build.did}" -a "{build.path}/{build.project_name}.iap"
```

Defragster, 2017-03-17:
Those two lines look just like two lines I already see in the platform.txt file?

In the IDE it runs a .BAT file after compiling that includes the script needed to perform the upload. After compile with Stino - I don't see any signs of that being run with the 'build option' 'show upload output' enabled.
`...\AppData\Local\Arduino15\packages\grumpyoldpizza\hardware\stm32l4\0.0.26\tools\windows\stm32l4-upload.bat`

I do see STM32 Bootloader under Programmer - and that doesn't work even when the device buttons are used to put it into the Bootloader mode that works from the IDE.

If you wanted to try you could install the JSON and build an example program and it will work - to fail the upload - even without having the board present. On my system I don't see it even trying to run the bat file to perform the upload.

Robot-Will, 2017-03-18:
I made an improvement, and now the upload commands are:

```
upload.pattern
"C:/Users/wood/AppData/Local/Arduino15/packages/grumpyoldpizza/hardware/stm32l4/0.0.26/tools/windows/stm32l4-upload.bat" "0x1209" "0x6669" "C:/Users/wood/AppData/Local/Arduino15/build/Blink/Blink.dfu"

program.pattern
"C:/Users/wood/AppData/Local/Arduino15/packages/grumpyoldpizza/hardware/stm32l4/0.0.26/tools/windows/stm32l4-upload.bat" "0x1209" "0x6669" "C:/Users/wood/AppData/Local/Arduino15/build/Blink/Blink.dfu"
```

Robot-Will, 2017-03-18:
There are two upload.pattern and program.pattern for DFU and OpenOCD!!!! I will check how to do with them. 

```
# Uploader tools
# --------------

# DFU
tools.stm32l4_dfu.cmd=stm32l4-upload
tools.stm32l4_dfu.cmd.windows=stm32l4-upload.bat
tools.stm32l4_dfu.path={runtime.platform.path}/tools/windows
tools.stm32l4_dfu.path.macosx={runtime.platform.path}/tools/macosx
tools.stm32l4_dfu.path.linux={runtime.platform.path}/tools/linux
tools.stm32l4_dfu.upload.params.verbose=-v
tools.stm32l4_dfu.upload.params.quiet=
tools.stm32l4_dfu.upload.pattern="{path}/{cmd}" "{build.vid}" "{build.pid}" "{build.path}/{build.project_name}.dfu"
tools.stm32l4_dfu.program.params.verbose=-v
tools.stm32l4_dfu.program.params.quiet=
tools.stm32l4_dfu.program.pattern="{path}/{cmd}" "{build.vid}" "{build.pid}" "{build.path}/{build.project_name}.dfu"
tools.stm32l4_dfu.erase.params.verbose=-v
tools.stm32l4_dfu.erase.params.quiet=
tools.stm32l4_dfu.erase.pattern=

# OpenOCD
tools.stm32l4_openocd.path={runtime.platform.path}/tools/windows/openocd/bin
tools.stm32l4_openocd.path.macosx={runtime.platform.path}/tools/macosx/openocd/bin
tools.stm32l4_openocd.path.linux={runtime.platform.path}/tools/linux/openocd/bin
tools.stm32l4_openocd.cmd=openocd
tools.stm32l4_openocd.cmd.windows=openocd.exe
tools.stm32l4_openocd.upload.params.verbose=-d2
tools.stm32l4_openocd.upload.params.quiet=-d0
tools.stm32l4_openocd.upload.pattern="{path}/{cmd}" {upload.verbose} -s "{runtime.platform.path}/tools/share/openocd/scripts/" -f "{runtime.platform.path}/variants/{build.variant}/{build.openocdscript}" -c "telnet_port disabled; program {{{build.path}/{build.project_name}.elf}} verify reset; shutdown"
tools.stm32l4_openocd.program.params.verbose=-d2
tools.stm32l4_openocd.program.params.quiet=-d0
tools.stm32l4_openocd.program.pattern="{path}/{cmd}" {program.verbose} -s "{runtime.platform.path}/tools/share/openocd/scripts/" -f "{runtime.platform.path}/variants/{build.variant}/{build.openocdscript}" -c "telnet_port disabled; program {{{build.path}/{build.project_name}.elf}} verify reset; shutdown"
tools.stm32l4_openocd.erase.params.verbose=-d3
tools.stm32l4_openocd.erase.params.quiet=-d0
tools.stm32l4_openocd.erase.pattern=
```

Defragster, 2017-03-18:
Thanks for looking so closely. Let me know if you have anything I can paste in and try on my end to platform.txt or other.

Defragster, 2017-03-18:
I just looked and see you are using '/' Forward slashes?  On windows it needs '\' back slashes to find the batch file like this is how I see mine execute for the upload:

C:\Users\xxx\AppData\Local\Arduino15\packages\grumpyoldpizza\hardware\stm32l4\0.0.26\tools\windows\stm32l4-upload.bat  0x1209  0x6669 C:\Users\xxxx\AppData\Local\Temp\arduino_build_381854/adcdma_t2.ino.dfu

There is one '/' - but that may be a param to a program that may correct it.

Also note the file itself is ...INO.dfu :: adcdma_t2.ino.dfu

Robot-Will, 2017-03-21:
I improved the upload commands. You can try it and see whether the commands work or not.

Defragster, 2017-03-22:
Branch moved - found the changes - if I have unsaved edits and do a REFORMAT the file is saved before - which is part of the new changes.

When I do UPLOAD I do not see any change- still no sign on the screen of any upload attempt.

Robot-Will, 2017-03-23:
The upload commands are:

```
"C:/Users/wood/AppData/Local/Arduino15/packages/grumpyoldpizza/hardware/stm32l4/0.0.26/tools/windows/stm32l4-upload.bat" "0x1209" "0x6669" "C:/Users/wood/AppData/Local/Arduino15/build/Blink/Blink.dfu"

"C:/Users/wood/AppData/Local/Arduino15/packages/grumpyoldpizza/hardware/stm32l4/0.0.26/tools/windows/openocd/bin/openocd.exe" -d0 -s "C:/Users/wood/AppData/Local/Arduino15/packages/grumpyoldpizza/hardware/stm32l4/0.0.26/tools/share/openocd/scripts/" -f "C:/Users/wood/AppData/Local/Arduino15/packages/grumpyoldpizza/hardware/stm32l4/0.0.26/variants/STM32L432KC-Ladybug/openocd_scripts/stm32l432kc_ladybug.cfg" -c "telnet_port disabled; program {{C:/Users/wood/AppData/Local/Arduino15/build/Blink/Blink.elf}} verify reset; shutdown"
```

Defragster, 2017-03-31:
Just getting back to this - is this something for me to try - or was there a change being pushed onto github?

Robot-Will, 2017-03-31:
You can try the latest version to see the upload process. Thanks.

And I added some codes to save current board settings for sketch, and once you save the settings for one sketch, when you active the sketch view, the saved settings will work. That is if you have different boards, and different sketches for diffrent boards, it will remember each board setting for each sketch.

Defragster, 2017-03-31:
I replaced the latest STINO in my Sublime packages and I'm not getting UPLOAD for either Teensy or STM32?

It stops with : "Build done."

Robot-Will, 2017-03-31:
Ctrl+` to see if there are error messages. Thanks.

Defragster, 2017-03-31:
I see this:

> plugins loaded
Package Control: Skipping automatic upgrade, last run at 2017-03-30 22:08:25, next run at 2017-03-30 23:08:25 or after
Forcing reset using 1200bps open/close on port COM6.
Couldn't find a Leonardo on the selected port. Check that you have the correct port selected. If it is correct, try pressing the board's reset button after initiating the upload.
Exception in thread Thread-24:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "C:\Users\Tim\AppData\Roaming\Sublime Text 3\Packages\Stino-Stino2017\libs\base_utils\task_queue.py", line 86, in _run
    self._consumer(*args)
  File "C:\Users\Tim\AppData\Roaming\Sublime Text 3\Packages\Stino-Stino2017\libs\stino_runtime\__init__.py", line 2162, in upload_sketch
    upload_cmd = upload_cmd.replace(upload_port, new_upload_port)
TypeError: Can't convert 'NoneType' object to str implicitly

Defragster, 2017-03-31:
For TEENSY it IS working to upload.

Though I see this in Ctrl+` window:

> plugins loaded
[Errno 2] No such file or directory: 'I:\\arduino-1.8.1\\hardware\\teensy\\avr\\programmers.txt'
Package Control: Skipping automatic upgrade, last run at 2017-03-30 22:08:25, next run at 2017-03-30 23:08:25 or after
[Errno 2] No such file or directory: 'I:\\arduino-1.8.1\\hardware\\teensy\\avr\\programmers.txt'
[Errno 2] No such file or directory: 'I:\\arduino-1.8.1\\hardware\\teensy\\avr\\programmers.txt'

Robot-Will, 2017-03-31:
I did something, you have another try, thanks.

Defragster, 2017-03-31:
IT WORKS!

However - it seems to be doing a BUILD ALL on each Upload - even with no changes made?

For STM32L4 - Updated Github/Stino, opened Sublime, it remembered settings for the INO, compiled and uploaded!
> Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
Upload done.

Defragster, 2017-03-31:
On Teensy also: It seems to always do a Arduino / Build Options / 'Full Build'

Perhaps that setting not setting from board defaults?

Also the STATUS bar on the bottom seems to often show the 'OTHER' board settings:
I have a file open for TEENSY and for STM32 and clicking the "Teensy sketch shows STM32 settings" and clicking the "STM32 sketch shows the Teensy settings"

Robot-Will, 2017-03-31:
It's my fault, I commented the file write function carelessly, so the settings did not save, and no last build info for build. It caused full build. I fixed it. Thanks.

Defragster, 2017-03-31:
Glad to help get it right.  Very good to see the progress and improvements!  

I inserted a couple of errors and I like how they are shown in the source file! And double clicking on the error shifting the code to view that error. Not sure if it would be nice to have the keyboard focus placed on that source line with the double click?

Robot-Will, 2017-03-31:
You mean the line in the stino output panel you click, and text cursor moves to the line?
No double click needed, just one click.

Defragster, 2017-03-31:
You are correct - it does it with a single click - I had my test errors too close to make it scroll.

When that is clicked - it could be nice to have the keyboard focus on that line for editing.

Robot-Will, 2017-03-31:
The error line in source file?

Defragster, 2017-03-31:
Yes, the error line in the source file is moved into view but the keyboard doesn't go there so I have to move the mouse and click to type.

Typically an error would be edited, if no editing is needed there the next error is still under the mouse and it can be clicked to select it so moving the keyboard focus should not bother in most cases.

Defragster, 2017-03-31:
Saving the board settings per sketch seems to allow doing multiple boards in the same session - this is very nice and did not work well before.

Robot-Will, 2017-03-31:
There are something wrong in Sublime Text, the source view can not be activated. If close the stino panel, the cursor is at the point of the error line. You can try to uncomment the 4 lines (158 - 161) in libs/stino_runtime/__init__.py, and when you click the error message line in stino panel, the panel closes, and the source view activated.

Defragster, 2017-03-31:
It looks very good now!  I can compile upload to Teensy and them STM32 and it works.
And when I switch back and forth a change to the INO file only recompiles a minimal subset and indicates the right board on the status bar and does a complete upload with prior settings.

I closed sublime and it does the same with a minimal rebuild and upload to the right board with proper settings.

That is fine with the error focus change - thanks for looking.

closed 2017-03-31 by Defragster