Robot-Will/Stino #382: 'Arduino / Install Platform / Add Arduino IDE' setting lot on restart

Defragster, 2017-03-16:
Even with just a single copy of Sublime open doing 'Arduino / Install Platform / Add Arduino IDE' setting is lost on Sublime restart?

Other JSON based boards are remembered - but to get to Teensy I have to Re-Add at least 4 times.

Also doesn't seem to restore the board I last used on that Platform on coming back.

Is this expected to work with multiple copies of Sublime open perhaps to different Boards/Platforms? It seems to revert to primary Console window and seemed confused.

Robot-Will, 2017-03-16:
1. I found a error and it causes the setting does not work, and I fixed it. 
2. Work with multiple copies of Sublime open perhaps to different Boards/Platforms? No.

Defragster, 2017-03-17:
The " Tools / Save All on Build " is not happening on Arduino Build or Upload.

Is that by any chance part of what was fixed above?

Robot-Will, 2017-03-17:
@Defragster  I will add this function.

Defragster, 2017-03-17:
Nice. At first I thought not saving was by design - then I saw that Sublime entry and realized it was an option.

BTW - I've done a few dozen compiles and uploads of sample sketches using Stino under Sublime and it is working very well with Teensy!

Robot-Will, 2017-03-17:
Done!

closed 2017-03-17 by Robot-Will