Robot-Will/Stino #383: gtest not found

mahluz, 2017-03-16:
im using sublime text 3 Arduino like IDE
and i got this error when compiling
Arduino/libraries/ArduinoJson/third-party/gtest-1.7.0/xcode/Samples/FrameworkSample/widget_test.cc:39:25: fatal error: gtest/gtest.h: No such file or directory
 #include "gtest/gtest.h"
but it successfully loaded on Arduino IDE

Robot-Will, 2017-03-17:
Please show me a test sketch, and I will find out why.

mahluz, 2017-03-17:
i just made a simple basicHttpClient using ESP8266,, help me please 
```
 `#include <Arduino.h>
 #include <ArduinoJson.h> // when i add this lib it give me an error
/*
Arduino/libraries/ArduinoJson/third-party/gtest-1.7.0/xcode/Samples/FrameworkSample/widget_test.cc:39:25: fatal error: gtest/gtest.h: No such file or directory
#include "gtest/gtest.h" 
i reinstall the library and then i got different error
Arduino/libraries/ArduinoJson/third-party/gtest-1.7.0/xcode/Samples/FrameworkSample/widget.cc:60:44: error: 'strncpy' was not declared in this scope
   strncpy(buffer, name_.c_str(), max_size-1);
*/
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>

ESP8266WiFiMulti WiFiMulti;

void setup() {
    Serial.begin(115200);
   // Serial.setDebugOutput(true);
    WiFiMulti.addAP("ssid", "pwd");
}

void loop() {
    // wait for WiFi connection
    if((WiFiMulti.run() == WL_CONNECTED)) {
        HTTPClient http;
        Serial.println("[HTTP] begin...");
        // configure traged server and url
        //http.begin("https://192.168.1.12/test.html", "7a 9c f4 db 40 d3 62 5a 6e 21 bc 5c cc 66 c8 3e a1 45 59 38"); //HTTPS
        http.begin("http://surveyunnes.esy.es/api/information"); //HTTP

        Serial.println("[HTTP] GET...");
        // start connection and send HTTP header
        int httpCode = http.GET();
        // httpCode will be negative on error
        if(httpCode > 0) {
         ``   // HTTP header has been send and Server response header has been handled
            Serial.printf("[HTTP] GET... code: %d\n", httpCode);

            // file found at server
            if(httpCode == HTTP_CODE_OK) {
                String json = http.getString();
                Serial.println(json);
            }
        } else {
            Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
        }
        http.end();
    }
    delay(2000);
}`
```

Robot-Will, 2017-03-17:
I fixed something, you can try it and see the message. If there are still errors, leave message to me. Thanks.

mahluz, 2017-03-18:
thankyou so much,, its working now.

closed 2017-03-18 by Robot-Will