Robot-Will/Stino #385: Auto Format on unsaved file uses last saved copy

Defragster, 2017-03-17:
Doing an Arduino Auto Format does not save the current file before working on the disk file copy.

Doing the Format loses any changes made. They come back with UNDO - but only if you notice.

Robot-Will, 2017-03-17:
Thanks alot. Save firstly! I will add the code.

Defragster, 2017-03-17:
New Sublime User Question:

How can I assign UPLOAD to Ctrl+u and Auto Format to Ctrl+t to execute the Stino commands?

instead of:
`	{ "keys": ["ctrl+u"], "command": "upper_case" },`
`	{ "keys": ["ctrl+t"], "command": "transpose" },`

Robot-Will, 2017-03-17:
Shortcuts is a big problem to satisfied all users around the world with different keyboards. So currently no shortcuts are defined. You can define your own shortcuts by your self. 

1. Open Setting File
Menu - Preferences - Package Settings - Stino - Key Bindings

2. Edit the file.
In Main.sublime-menu you can find all commands. Build is stino_build, upload is stino_upload. Then write the Key bindings as SublimeText documents.

3. Save.

Thanks.

Defragster, 2017-03-17:
Got it working - I looked in the wrong file for the stino_ command names.

[
{ "keys": ["ctrl+u"], "command": "stino_upload" },
{ "keys": ["ctrl+t"], "command": "stino_auto_format" }
]

Thanks!  I'm glad you have this working so well now!  It was easy to setup and get working and is much better than the IDE interface - but I like that it uses the Arduino install because it let you get it working for Teensy setup.

Robot-Will, 2017-03-17:
Thanks a lot. I added the save command.

closed 2017-03-17 by Robot-Will