Robot-Will/Stino #386: Problems with Sublime Text and the Arduino pack "Stino2017".

ciprus, 2017-03-17:
Hello,

I have problems with ST (Sublime Text) and the Arduino pack "Stino2017".
[image]
[image]

I am under W10 in 64 bits.
Here is the installation procedure I have done:

Installation in order:
- Arduino 1.8.1 (and operation)
- ST 3126
Place the file:
- "Package Control.sublime-package" in: C: \ Program Files (x86) / Sublime Text3 / Packages
PackageControl Appears Well in ST's Preferences
- Stino2017 in: user (s) / AppData / Roaming / Sublime Text3 / Packages by uncopier & paste from the "Stino2017" folder in Preferences / BrowsePackages.

Then I did: Preferences / PackagesControl / Install Package and then install "Arduino Like".
I get the Arduino menu but not with the right information as the preferences of the Arduino do not appear.

Robot-Will, 2017-03-17:
What's wrong? Just install your platform from internet or add a Arduino IDE on your HDD, then use it.

ciprus, 2017-03-17:
Hello and thank you for your answer.
As I said in my post, I installed the Arduino software and it works.
My problem as you can see from the images I attached to my post in the Arduino menu pane, I do not have the menu for preferences so I can not link to Sublime Text and the software Arduino; If I click for example on language, I have the refrech and install information that appears but is grayed out and I can not click on it. It's the same for "board option etc."

Robot-Will, 2017-03-17:
1. Do not do this step : Then I did: Preferences / PackagesControl / Install Package and then install "Arduino Like".

2. This plugin has nothing to do with Arduino IDE, they are two totally different softwares.

3. Menu -> Arduino -> Install Platform -> Add Arduino IDE

ciprus, 2017-03-17:
Well, I completely uninstalled Sublime Text3.
I reinstalled Sublime Text3.
I put Stino2017 in the package folder "Browse Package".
I restarted Sublime Text3.
I clicked on the "Arduino" menu of S.T.3, then "install Platform / Add Arduino IDE".
I restarted S.T.3., Nothing has changed, I still do not have the menu preference in "Arduino"
[image]

Robot-Will, 2017-03-17:
1. How did you install ST？ exe or zip? You should open ST packages folder: menu - Preferences - packages

Then delete Arduino Like folder If it exists.

2. Do you have a internet connection? In menu Install platform there will be a list of avilabale platforms after download package index files.

3. Did you input the correct path of Arduino IDE?

4.Or there are errors on your pc. So press Ctrl+` to open ST's console to see the error messages and show them to me.

ciprus, 2017-03-18:
Hello,
- I deleted and reinstalled ST to restart on a good basis.
- I copied the folder "Stino2017 in" Browse packages "
- I restarted ST
- I installed ST in "exe" (Sublime Text Build 3126 Setup.exe)
- I check in "preferences / packagesControl / Disable Package, there is not the file" Arduino like ".
- For the "Install platform" menu, I have the list that appears on the image that I sent you in my last message. I have no other list of "platforms"
- I do not have the correct path of the Arduino, because in the Arduino menu, I do not have the "preferences" line to indicate the path to the Arduino software.

DPI scale: 1
startup, version: 3126 windows x32 channel: stable
executable: /C/Program Files (x86)/Sublime Text 3/sublime_text.exe
working dir: /C/Program Files (x86)/Sublime Text 3
packages path: /C/Users/PC-WEB/AppData/Roaming/Sublime Text 3/Packages
state path: /C/Users/PC-WEB/AppData/Roaming/Sublime Text 3/Local
zip path: /C/Program Files (x86)/Sublime Text 3/Packages
zip path: /C/Users/PC-WEB/AppData/Roaming/Sublime Text 3/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 0.409141
startup time: 0.456141
first paint time: 0.456141
first paint time: 0.456141
Unable to fetch update url contents
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin Stino-Stino2017.StinoCommands
plugins loaded
http://downloads.arduino.cc/libraries/library_index.json <urlopen error [Errno 11001] getaddrinfo failed>
http://downloads.arduino.cc/packages/package_index.json <urlopen error [Errno 11001] getaddrinfo failed>

Robot-Will, 2017-03-18:
Your internet connection is not good. Download of index files failed.

```
http://downloads.arduino.cc/libraries/library_index.json <urlopen error [Errno 11001] getaddrinfo failed>
http://downloads.arduino.cc/packages/package_index.json <urlopen error [Errno 11001] getaddrinfo failed>
```

Robot-Will, 2017-03-18:
Add your Arduino IDE:

[image]

[image]

ciprus, 2017-03-19:
Hello
I managed to move forward in the installation of ST3, my firewall blocked the connection to Stino 2017.
I copied Stino 2017 in preferences / browsepackage.
Then I made Arduino / Install Platform / Add Arduino IDE, I made the connection between Arduino 1.8.1 and ST3 in IDE Arduno path: c: \ program files (x86) \ Arduino after I get blocked because I Does not make the connection between ST3 and Arduino.
I attach the file given by the console as well as two images. Thank you

DPI scale: 1
startup, version: 3126 windows x32 channel: stable
executable: /C/Program Files (x86)/Sublime Text 3/sublime_text.exe
working dir: /C/Program Files (x86)/Sublime Text 3
packages path: /C/Users/PC-WEB/AppData/Roaming/Sublime Text 3/Packages
state path: /C/Users/PC-WEB/AppData/Roaming/Sublime Text 3/Local
zip path: /C/Program Files (x86)/Sublime Text 3/Packages
zip path: /C/Users/PC-WEB/AppData/Roaming/Sublime Text 3/Installed Packages
ignored_packages: ["Vintage"]
pre session restore time: 1.23239
startup time: 1.34239
first paint time: 1.34239
first paint time: 1.34239
reloading plugin Default.auto_indent_tag
reloading plugin Default.block
reloading plugin Default.comment
reloading plugin Default.convert_syntax
reloading plugin Default.copy_path
reloading plugin Default.delete_word
reloading plugin Default.detect_indentation
reloading plugin Default.duplicate_line
reloading plugin Default.echo
reloading plugin Default.exec
reloading plugin Default.fold
reloading plugin Default.font
reloading plugin Default.goto_line
reloading plugin Default.history_list
reloading plugin Default.indentation
reloading plugin Default.install_package_control
reloading plugin Default.kill_ring
reloading plugin Default.mark
reloading plugin Default.new_templates
reloading plugin Default.open_context_url
reloading plugin Default.open_in_browser
reloading plugin Default.pane
reloading plugin Default.paragraph
reloading plugin Default.paste_from_history
reloading plugin Default.profile
reloading plugin Default.quick_panel
reloading plugin Default.run_syntax_tests
reloading plugin Default.save_on_focus_lost
reloading plugin Default.scroll
reloading plugin Default.set_unsaved_view_name
reloading plugin Default.settings
reloading plugin Default.show_scope_name
reloading plugin Default.side_bar
reloading plugin Default.sort
reloading plugin Default.swap_line
reloading plugin Default.switch_file
reloading plugin Default.symbol
reloading plugin Default.transform
reloading plugin Default.transpose
reloading plugin Default.trim_trailing_white_space
reloading plugin CSS.css_completions
reloading plugin Diff.diff
reloading plugin HTML.encode_html_entities
reloading plugin HTML.html_completions
reloading plugin Stino-Stino2017.StinoCommands
plugins loaded

[image]
[image]

Robot-Will, 2017-03-21:
I really do not know what is your problem. Input the path and you will find the platforms in Arduino IDE.

[image]
[image]

closed 2017-03-21 by Robot-Will

Robot-Will, 2017-03-23:
If you can not see the contents of all menus, just do the following steps:

1. Click:   Menu -> Preferences -> Settings

2. In the right view (User), delete all contents in "ignored_packages".

Thanks.