Robot-Will/Stino #387: #include inside #ifdef includes library...

KurtE, 2017-03-17:
I have one project where I may or may not build with an IMU.  So the project has code like:
```
#ifdef USE_LSM9DS1
#include <Wire.h>
#include <SPI.h>
#include <SparkFunLSM9DS1.h>
...
```
Normally I don't have this define defined in my program so under Arduino builds it is not included as part of the project.   (https://github.com/KurtE/Teensy_Arbotix_Pro)

However with current Stino2017,  it is including the source files of that library where I am seeing compiler warnings: 
```
Build] C:\Users\kurte\Documents\Arduino\Github\Teensy_Arbotix_Pro...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[1.4%] Compiling SparkFunLSM9DS1.cpp...
C:/Users/kurte/Documents/Arduino/libraries/SparkFunLSM9DS1/src/SparkFunLSM9DS1.cpp: In member function 'void LSM9DS1::calibrate(bool)':
C:/Users/kurte/Documents/Arduino/libraries/SparkFunLSM9DS1/src/SparkFunLSM9DS1.cpp:328:10: warning: unused variable 'data' [-Wunused-variable]
  uint8_t data[6] = {0, 0, 0, 0, 0, 0};
          ^
C:/Users/kurte/Documents/Arduino/libraries/SparkFunLSM9DS1/src/SparkFunLSM9DS1.cpp: In member function 'uint8_t LSM9DS1::I2CreadBytes(uint8_t, uint8_t, uint8_t*, uint8_t)':
C:/Users/kurte/Documents/Arduino/libraries/SparkFunLSM9DS1/src/SparkFunLSM9DS1.cpp:1132:10: warning: unused variable 'i' [-Wunused-variable]
  uint8_t i = 0;
```

This is probably not a high priority as I will probably be cutting this code out anyway... Trying out different IMU boards, but though I would mention it.

Robot-Will, 2017-03-18:
I downloaded this project(https://github.com/KurtE/Teensy_Arbotix_Pro), and build it, without any warning messages. Or not this project?

```
[Build] D:\Documents\Arduino\Teensy_Arbotix_Pro-master...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[1.5%] Compiling SPI.cpp...
[2.9%] Compiling twi.c...
[4.4%] Compiling Wire.cpp...
[5.9%] Compiling esp8266.c...
[7.4%] Compiling Adafruit_NeoPixel.cpp...
[8.8%] Compiling USBInput.cpp...
[10.3%] Compiling SyncRead.cpp...
[11.8%] Compiling LocalRegisters.cpp...
[13.2%] Compiling imu_Sparkfun_LSM9DS1.cpp...
[14.7%] Compiling AXBuss.cpp...
[16.2%] Compiling BioloidSerial.cpp...
[17.6%] Compiling ax12Serial.cpp...
[19.1%] Compiling memset.S...
[20.6%] Compiling memcpy-armv7m.S...
[22.1%] Compiling usb_touch.c...
[23.5%] Compiling usb_serial.c...
[25.0%] Compiling usb_seremu.c...
[26.5%] Compiling usb_rawhid.c...
[27.9%] Compiling usb_mtp.c...
[29.4%] Compiling usb_mouse.c...
[30.9%] Compiling usb_midi.c...
[32.4%] Compiling usb_mem.c...
[33.8%] Compiling usb_keyboard.c...
[35.3%] Compiling usb_joystick.c...
[36.8%] Compiling usb_dev.c...
[38.2%] Compiling usb_desc.c...
[39.7%] Compiling touch.c...
[41.2%] Compiling ser_print.c...
[42.6%] Compiling serial6_lpuart.c...
[44.1%] Compiling serial6.c...
[45.6%] Compiling serial5.c...
[47.1%] Compiling serial4.c...
[48.5%] Compiling serial3.c...
[50.0%] Compiling serial2.c...
[51.5%] Compiling serial1.c...
[52.9%] Compiling pins_teensy.c...
[54.4%] Compiling nonstd.c...
[55.9%] Compiling mk20dx128.c...
[57.4%] Compiling math_helper.c...
[58.8%] Compiling keylayouts.c...
[60.3%] Compiling eeprom.c...
[61.8%] Compiling analog.c...
[63.2%] Compiling yield.cpp...
[64.7%] Compiling WString.cpp...
[66.2%] Compiling WMath.cpp...
[67.6%] Compiling usb_inst.cpp...
[69.1%] Compiling usb_flightsim.cpp...
[70.6%] Compiling usb_audio.cpp...
[72.1%] Compiling Tone.cpp...
[73.5%] Compiling Stream.cpp...
[75.0%] Compiling Print.cpp...
[76.5%] Compiling new.cpp...
[77.9%] Compiling main.cpp...
[79.4%] Compiling IPAddress.cpp...
[80.9%] Compiling IntervalTimer.cpp...
[82.4%] Compiling HardwareSerial6.cpp...
[83.8%] Compiling HardwareSerial5.cpp...
[85.3%] Compiling HardwareSerial4.cpp...
[86.8%] Compiling HardwareSerial3.cpp...
[88.2%] Compiling HardwareSerial2.cpp...
[89.7%] Compiling HardwareSerial1.cpp...
[91.2%] Compiling DMAChannel.cpp...
[92.6%] Compiling avr_emulation.cpp...
[94.1%] Compiling AudioStream.cpp...
[95.6%] Compiling EEPROM.cpp...
[97.1%] Compiling Teensy_Arbotix_Pro-master.ino.cpp...
[98.5%] Linking everything together...
[100.0%] Creating binary files...
Opening Teensy Loader...
Sketch uses 30,440 bytes (2.9%) of program storage space. Maximum is 1,048,576 bytes.
Global variables use 5,432 bytes (2.1%) of dynamic memory, leaving 256,712 bytes for local variables. Maximum is 262,144 bytes.
```

Robot-Will, 2017-03-18:
Which board and board options?

KurtE, 2017-03-18:
I was building for Teensy 3.6.  
As for no compiler warnings... My guess is Sparkfun, maybe updated their library since I used it. 

But I guess the question is, Do you have the SparkFunLSM9DS1 library installed on your machine? 
Again in this case it may not be that important, but later I may have code that does stuff like: 
```
#ifdef USE_I2C_T3
#include <i2c_t3.h>
#else
#include <wire.h>
#endif
```

I have not built a program with this yet, but if both libraries are included in the build they will fail as both create and use same objects.   Will investigate more.

Thanks

Robot-Will, 2017-03-20:
I installed the SparkFun LSM9DS1 library and I add the following codes and got no errors.

```
#ifdef USE_I2C_T3
#include <i2c_t3.h>
#else
#include <wire.h>
#endif
```

Then I added all .h to the project, and got the warnings of SparkFunLSM9DS1.cpp. I think the reason is something wrong in this cpp file. The messages are given by the compiler (g++), not this plugin.

```
//=============================================================================
//Project Teensy Arbotix Pro
//Description: To have an Teensy 3.1/3.2, semi emulate an Arbotix Pro or
//      Robotis CM730
//=============================================================================

//=============================================================================
// Header Files
//=============================================================================

#include <EEPROM.h>

#include <ax12Serial.h>
#include <BioloidSerial.h>
#include "globals.h"
#include "imu.h"

#if NEOPIXEL_PIN
#include <Adafruit_NeoPixel.h>
//   NEO_KHZ800  800 KHz bitstream (most NeoPixel products w/WS2812 LEDs)
//   NEO_KHZ400  400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
//   NEO_GRB     Pixels are wired for GRB bitstream (most NeoPixel products)
//   NEO_RGB     Pixels are wired for RGB bitstream (v1 FLORA pixels, not v2)
Adafruit_NeoPixel strip = Adafruit_NeoPixel(1, NEOPIXEL_PIN, NEO_RGB + NEO_KHZ800);
#endif

#include <Adafruit_NeoPixel.h>
#include <i2c_t3.h>
#include <SPI.h>
#include <SparkFunLSM9DS1.h>

//=============================================================================
//[CONSTANTS]
//=============================================================================
......
```

```
[Build] D:\Documents\Arduino\Teensy_Arbotix_Pro-master...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[1.4%] Compiling i2c_t3.cpp...
[2.9%] Compiling SparkFunLSM9DS1.cpp...
D:/Documents/Arduino/libraries/SparkFun_LSM9DS1_Arduino_Library-master/src/SparkFunLSM9DS1.cpp: In member function 'void LSM9DS1::calibrate(bool)':
D:/Documents/Arduino/libraries/SparkFun_LSM9DS1_Arduino_Library-master/src/SparkFunLSM9DS1.cpp:339:10: warning: unused variable 'data' [-Wunused-variable]
  uint8_t data[6] = {0, 0, 0, 0, 0, 0};
          ^
D:/Documents/Arduino/libraries/SparkFun_LSM9DS1_Arduino_Library-master/src/SparkFunLSM9DS1.cpp: In member function 'uint8_t LSM9DS1::xgReadByte(uint8_t)':
D:/Documents/Arduino/libraries/SparkFun_LSM9DS1_Arduino_Library-master/src/SparkFunLSM9DS1.cpp:1055:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
D:/Documents/Arduino/libraries/SparkFun_LSM9DS1_Arduino_Library-master/src/SparkFunLSM9DS1.cpp: In member function 'uint8_t LSM9DS1::xgReadBytes(uint8_t, uint8_t*, uint8_t)':
D:/Documents/Arduino/libraries/SparkFun_LSM9DS1_Arduino_Library-master/src/SparkFunLSM9DS1.cpp:1065:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
D:/Documents/Arduino/libraries/SparkFun_LSM9DS1_Arduino_Library-master/src/SparkFunLSM9DS1.cpp: In member function 'uint8_t LSM9DS1::mReadByte(uint8_t)':
[... 55 lines cut ...]
[72.9%] Compiling Tone.cpp...
[74.3%] Compiling Stream.cpp...
[75.7%] Compiling Print.cpp...
[77.1%] Compiling new.cpp...
[78.6%] Compiling main.cpp...
[80.0%] Compiling IPAddress.cpp...
[81.4%] Compiling IntervalTimer.cpp...
[82.9%] Compiling HardwareSerial6.cpp...
[84.3%] Compiling HardwareSerial5.cpp...
[85.7%] Compiling HardwareSerial4.cpp...
[87.1%] Compiling HardwareSerial3.cpp...
[88.6%] Compiling HardwareSerial2.cpp...
[90.0%] Compiling HardwareSerial1.cpp...
[91.4%] Compiling DMAChannel.cpp...
[92.9%] Compiling avr_emulation.cpp...
[94.3%] Compiling AudioStream.cpp...
[95.7%] Compiling EEPROM.cpp...
[97.1%] Compiling Teensy_Arbotix_Pro-master.ino.cpp...
[98.6%] Linking everything together...
[100.0%] Creating binary files...
Opening Teensy Loader...
Sketch uses 34,928 bytes (3.3%) of program storage space. Maximum is 1,048,576 bytes.
Global variables use 7,984 bytes (3.0%) of dynamic memory, leaving 254,160 bytes for local variables. Maximum is 262,144 bytes.

```

KurtE, 2017-03-20:
Maybe it would help to make a simple test case... Suppose I have the following sketch file:
```
#ifdef USE_BAD_LIBRARY
#include <BadLibrary.h>
#endif

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

And I create dummy library (arduino folder BadLibrary
I have the two files:
BadLibrary.h
```
#ifndef BAD_LIBRARY
#define BAD_LIBRARY
typdef struct some_struct {
	int some_var;
} SOME_STRUCT;
#endif
```

and BadLibrary.cpp
```
#include "BadLibrary.h"
This is bogus
'''

If I build under Arduino it compiles without error as it does not include the library
```

Opening Teensy Loader...

Sketch uses 14516 bytes (5%) of program storage space. Maximum is 262144 bytes.
Global variables use 3480 bytes (5%) of dynamic memory, leaving 62056 bytes for local variables. Maximum is 65536 bytes.
```

However when I try to build with Stino it fails as it includes the library into the build including the bogus .cpp file
```
[Build] C:\Users\kurte\Documents\Arduino\BadLIbraryTest...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[1.8%] Compiling memset.S...
[3.6%] Compiling memcpy-armv7m.S...
[5.4%] Compiling usb_touch.c...
[7.1%] Compiling usb_serial.c...
[8.9%] Compiling usb_seremu.c...
[10.7%] Compiling usb_rawhid.c...
[12.5%] Compiling usb_mtp.c...
[14.3%] Compiling usb_mouse.c...
[16.1%] Compiling usb_midi.c...
[17.9%] Compiling usb_mem.c...
[19.6%] Compiling usb_keyboard.c...
[21.4%] Compiling usb_joystick.c...
[23.2%] Compiling usb_dev.c...
[25.0%] Compiling usb_desc.c...
[26.8%] Compiling touch.c...
[28.6%] Compiling ser_print.c...
[30.4%] Compiling serial6_lpuart.c...
[32.1%] Compiling serial6.c...
[33.9%] Compiling serial5.c...
[35.7%] Compiling serial4.c...
[37.5%] Compiling serial3.c...
[39.3%] Compiling serial2.c...
[41.1%] Compiling serial1.c...
[42.9%] Compiling pins_teensy.c...
[44.6%] Compiling nonstd.c...
[46.4%] Compiling mk20dx128.c...
[48.2%] Compiling math_helper.c...
[50.0%] Compiling keylayouts.c...
[51.8%] Compiling eeprom.c...
[53.6%] Compiling analog.c...
[55.4%] Compiling yield.cpp...
[57.1%] Compiling WString.cpp...
[58.9%] Compiling WMath.cpp...
[60.7%] Compiling usb_inst.cpp...
[62.5%] Compiling usb_flightsim.cpp...
[64.3%] Compiling usb_audio.cpp...
[66.1%] Compiling Tone.cpp...
[67.9%] Compiling Stream.cpp...
[69.6%] Compiling Print.cpp...
[71.4%] Compiling new.cpp...
[73.2%] Compiling main.cpp...
[75.0%] Compiling IPAddress.cpp...
[76.8%] Compiling IntervalTimer.cpp...
[78.6%] Compiling HardwareSerial6.cpp...
[80.4%] Compiling HardwareSerial5.cpp...
[82.1%] Compiling HardwareSerial4.cpp...
[83.9%] Compiling HardwareSerial3.cpp...
[85.7%] Compiling HardwareSerial2.cpp...
[87.5%] Compiling HardwareSerial1.cpp...
[89.3%] Compiling DMAChannel.cpp...
[91.1%] Compiling avr_emulation.cpp...
[92.9%] Compiling AudioStream.cpp...
[94.6%] Compiling BadLibrary.cpp...
In file included from C:/Users/kurte/Documents/Arduino/libraries/BadLibrary/BadLibrary.cpp:1:0:
C:/Users/kurte/Documents/Arduino/libraries/BadLibrary/BadLibrary.h:3:1: error: 'typdef' does not name a type
 typdef struct some_struct {
 ^
C:/Users/kurte/Documents/Arduino/libraries/BadLibrary/BadLibrary.h:5:3: error: 'SOME_STRUCT' does not name a type
 } SOME_STRUCT;
   ^
C:/Users/kurte/Documents/Arduino/libraries/BadLibrary/BadLibrary.cpp:2:1: error: 'This' does not name a type
 This is bogus
 ^
```

Robot-Will, 2017-03-21:
Arduino IDE 1.8.1, same errors. 

```
E:\dev\arduino\arduino-1.8.1\arduino-builder -dump-prefs -logger=machine -hardware E:\dev\arduino\arduino-1.8.1\hardware -hardware C:\Users\wood\AppData\Local\Arduino15\packages -hardware D:\Documents\Arduino\hardware -tools E:\dev\arduino\arduino-1.8.1\tools-builder -tools E:\dev\arduino\arduino-1.8.1\hardware\tools\avr -tools C:\Users\wood\AppData\Local\Arduino15\packages -built-in-libraries E:\dev\arduino\arduino-1.8.1\libraries -libraries D:\Documents\Arduino\libraries -fqbn=teensy:avr:teensy36:usb=serial,speed=180,opt=o1std,keys=en-us -ide-version=10801 -build-path C:\Users\wood\AppData\Local\Temp\arduino_build_659802 -warnings=none -verbose D:\Documents\Arduino\macro_test\macro_test.ino
E:\dev\arduino\arduino-1.8.1\arduino-builder -compile -logger=machine -hardware E:\dev\arduino\arduino-1.8.1\hardware -hardware C:\Users\wood\AppData\Local\Arduino15\packages -hardware D:\Documents\Arduino\hardware -tools E:\dev\arduino\arduino-1.8.1\tools-builder -tools E:\dev\arduino\arduino-1.8.1\hardware\tools\avr -tools C:\Users\wood\AppData\Local\Arduino15\packages -built-in-libraries E:\dev\arduino\arduino-1.8.1\libraries -libraries D:\Documents\Arduino\libraries -fqbn=teensy:avr:teensy36:usb=serial,speed=180,opt=o1std,keys=en-us -ide-version=10801 -build-path C:\Users\wood\AppData\Local\Temp\arduino_build_659802 -warnings=none -verbose D:\Documents\Arduino\macro_test\macro_test.ino
Using board 'teensy36' from platform in folder: E:\dev\arduino\arduino-1.8.1\hardware\teensy\avr
Using core 'teensy3' from platform in folder: E:\dev\arduino\arduino-1.8.1\hardware\teensy\avr
Detecting libraries used...
"E:\dev\arduino\arduino-1.8.1\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=134 -DARDUINO=10801 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IE:\dev\arduino\arduino-1.8.1\hardware\teensy\avr\cores\teensy3" "C:\Users\wood\AppData\Local\Temp\arduino_build_659802\sketch\macro_test.ino.cpp" -o "nul"
"E:\dev\arduino\arduino-1.8.1\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=134 -DARDUINO=10801 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IE:\dev\arduino\arduino-1.8.1\hardware\teensy\avr\cores\teensy3" "C:\Users\wood\AppData\Local\Temp\arduino_build_659802\sketch\BadLibrary.cpp" -o "nul"
Generating function prototypes...
"E:\dev\arduino\arduino-1.8.1\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=134 -DARDUINO=10801 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IE:\dev\arduino\arduino-1.8.1\hardware\teensy\avr\cores\teensy3" "C:\Users\wood\AppData\Local\Temp\arduino_build_659802\sketch\macro_test.ino.cpp" -o "C:\Users\wood\AppData\Local\Temp\arduino_build_659802\preproc\ctags_target_for_gcc_minus_e.cpp"
"E:\dev\arduino\arduino-1.8.1\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\wood\AppData\Local\Temp\arduino_build_659802\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"E:\dev\arduino\arduino-1.8.1\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -c -O1 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=134 -DARDUINO=10801 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IE:\dev\arduino\arduino-1.8.1\hardware\teensy\avr\cores\teensy3" "C:\Users\wood\AppData\Local\Temp\arduino_build_659802\sketch\BadLibrary.cpp" -o "C:\Users\wood\AppData\Local\Temp\arduino_build_659802\sketch\BadLibrary.cpp.o"
In file included from C:\Users\wood\AppData\Local\Temp\arduino_build_659802\sketch\BadLibrary.cpp:1:0:

BadLibrary.h:3: error: 'typdef' does not name a type
 typdef struct some_struct {

 ^

BadLibrary.h:5: error: 'SOME_STRUCT' does not name a type
 } SOME_STRUCT;

   ^

'typdef' does not name a type

```

KurtE, 2017-03-21:
Sort of interesting as I just retried with 1.8.1 with TD 1.36-beta3...
```

D:\arduino-1.8.1\arduino-builder -dump-prefs -logger=machine -hardware D:\arduino-1.8.1\hardware -hardware C:\Users\kurte\AppData\Local\Arduino15\packages -hardware C:\Users\kurte\Documents\Arduino\hardware -tools D:\arduino-1.8.1\tools-builder -tools D:\arduino-1.8.1\hardware\tools\avr -tools C:\Users\kurte\AppData\Local\Arduino15\packages -built-in-libraries D:\arduino-1.8.1\libraries -libraries C:\Users\kurte\Documents\Arduino\libraries -fqbn=teensy:avr:teensy31:usb=serial,speed=96,opt=o2std,keys=en-us -ide-version=10801 -build-path C:\Users\kurte\AppData\Local\Temp\arduino_build_228645 -warnings=none -verbose C:\Users\kurte\Documents\Arduino\BadLIbraryTest\BadLIbraryTest.ino
D:\arduino-1.8.1\arduino-builder -compile -logger=machine -hardware D:\arduino-1.8.1\hardware -hardware C:\Users\kurte\AppData\Local\Arduino15\packages -hardware C:\Users\kurte\Documents\Arduino\hardware -tools D:\arduino-1.8.1\tools-builder -tools D:\arduino-1.8.1\hardware\tools\avr -tools C:\Users\kurte\AppData\Local\Arduino15\packages -built-in-libraries D:\arduino-1.8.1\libraries -libraries C:\Users\kurte\Documents\Arduino\libraries -fqbn=teensy:avr:teensy31:usb=serial,speed=96,opt=o2std,keys=en-us -ide-version=10801 -build-path C:\Users\kurte\AppData\Local\Temp\arduino_build_228645 -warnings=none -verbose C:\Users\kurte\Documents\Arduino\BadLIbraryTest\BadLIbraryTest.ino
Using board 'teensy31' from platform in folder: D:\arduino-1.8.1\hardware\teensy\avr
Using core 'teensy3' from platform in folder: D:\arduino-1.8.1\hardware\teensy\avr
Detecting libraries used...
"D:\arduino-1.8.1\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=136 -DARDUINO=10801 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-ID:\arduino-1.8.1\hardware\teensy\avr\cores\teensy3" "C:\Users\kurte\AppData\Local\Temp\arduino_build_228645\sketch\BadLIbraryTest.ino.cpp" -o "nul"
Generating function prototypes...
"D:\arduino-1.8.1\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=136 -DARDUINO=10801 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-ID:\arduino-1.8.1\hardware\teensy\avr\cores\teensy3" "C:\Users\kurte\AppData\Local\Temp\arduino_build_228645\sketch\BadLIbraryTest.ino.cpp" -o "C:\Users\kurte\AppData\Local\Temp\arduino_build_228645\preproc\ctags_target_for_gcc_minus_e.cpp"
"D:\arduino-1.8.1\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\kurte\AppData\Local\Temp\arduino_build_228645\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"D:\arduino-1.8.1\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -c -O2 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=136 -DARDUINO=10801 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-ID:\arduino-1.8.1\hardware\teensy\avr\cores\teensy3" "C:\Users\kurte\AppData\Local\Temp\arduino_build_228645\sketch\BadLIbraryTest.ino.cpp" -o "C:\Users\kurte\AppData\Local\Temp\arduino_build_228645\sketch\BadLIbraryTest.ino.cpp.o"
Compiling libraries...
Compiling core...
"D:\arduino-1.8.1\hardware\teensy/../tools/arm/bin/arm-none-eabi-gcc" -c -O2 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -x assembler-with-cpp -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=136 -DARDUINO=10801 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-ID:\arduino-1.8.1\hardware\teensy\avr\cores\teensy3" "D:\arduino-1.8.1\hardware\teensy\avr\cores\teensy3\memcpy-armv7m.S" -o "C:\Users\kurte\AppData\Local\Temp\arduino_build_228645\core\memcpy-armv7m.S.o"
"D:\arduino-1.8.1\hardware\teensy/../tools/arm/bin/arm-none-eabi-gcc" -c -O2 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -x assembler-with-cpp -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=136 -DARDUINO=10801 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-ID:\arduino-1.8.1\hardware\teensy\avr\cores\teensy3" "D:\arduino-1.8.1\hardware\teensy\avr\cores\teensy3\memset.S" -o "C:\Users\kurte\AppData\Local\Temp\arduino_build_228645\core\memset.S.o"
"D:\arduino-1.8.1\hardware\teensy/../tools/arm/bin/arm-none-eabi-gcc" -c -O2 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD  -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=136 -DARDUINO=10801 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-ID:\arduino-1.8.1\hardware\teensy\avr\cores\teensy3" "D:\arduino-1.8.1\hardware\teensy\avr\cores\teensy3\analog.c" -o "C:\Users\kurte\AppData\Local\Temp\arduino_build_228645\core\analog.c.o"
"D:\arduino-1.8.1\hardware\teensy/../tools/arm/bin/arm-none-eabi-gcc" -c -O2 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD  -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=136 -DARDUINO=10801 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-ID:\arduino-1.8.1\hardware\teensy\avr\ [... cut ...]
Don't know what the difference is.  Maybe because I created the Sketch and library, and then exited Arduino and then restarted Arduino...  Sometimes in past found that if I used a library for awhile in a sketch even if I remove the reference to it, it will continue to include the library during that session.  So starting it fresh has the system again scan for what it thinks it needs... 

Again maybe not the highest priority issue.  I would much rather have things like being able to click no the error message open up the file and position to the error and other such features.

Robot-Will, 2017-03-22:
Error Message I added yesterday, you can try it. Still there are some features not done, I am still working on it. Thanks.

Robot-Will, 2017-03-23:
Arduino IDE 1.8.2 + Teensy 1.36b4, I can not pass the build, same errors.

```
E:\dev\arduino\arduino-1.8.2\arduino-builder -dump-prefs -logger=machine -hardware E:\dev\arduino\arduino-1.8.2\hardware -hardware C:\Users\wood\AppData\Local\Arduino15\packages -hardware D:\Documents\Arduino\hardware -tools E:\dev\arduino\arduino-1.8.2\tools-builder -tools E:\dev\arduino\arduino-1.8.2\hardware\tools\avr -tools C:\Users\wood\AppData\Local\Arduino15\packages -built-in-libraries E:\dev\arduino\arduino-1.8.2\libraries -libraries D:\Documents\Arduino\libraries -fqbn=teensy:avr:teensy36:usb=serial,speed=180,opt=o1std,keys=en-us -vid-pid=0X2341_0X0010 -ide-version=10802 -build-path C:\Users\wood\AppData\Local\Temp\arduino_build_863737 -warnings=none -verbose D:\Documents\Arduino\macro_test\macro_test.ino
E:\dev\arduino\arduino-1.8.2\arduino-builder -compile -logger=machine -hardware E:\dev\arduino\arduino-1.8.2\hardware -hardware C:\Users\wood\AppData\Local\Arduino15\packages -hardware D:\Documents\Arduino\hardware -tools E:\dev\arduino\arduino-1.8.2\tools-builder -tools E:\dev\arduino\arduino-1.8.2\hardware\tools\avr -tools C:\Users\wood\AppData\Local\Arduino15\packages -built-in-libraries E:\dev\arduino\arduino-1.8.2\libraries -libraries D:\Documents\Arduino\libraries -fqbn=teensy:avr:teensy36:usb=serial,speed=180,opt=o1std,keys=en-us -vid-pid=0X2341_0X0010 -ide-version=10802 -build-path C:\Users\wood\AppData\Local\Temp\arduino_build_863737 -warnings=none -verbose D:\Documents\Arduino\macro_test\macro_test.ino
Using board 'teensy36' from platform in folder: E:\dev\arduino\arduino-1.8.2\hardware\teensy\avr
Using core 'teensy3' from platform in folder: E:\dev\arduino\arduino-1.8.2\hardware\teensy\avr
Detecting libraries used...
"E:\dev\arduino\arduino-1.8.2\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=136 -DARDUINO=10802 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IE:\dev\arduino\arduino-1.8.2\hardware\teensy\avr\cores\teensy3" "C:\Users\wood\AppData\Local\Temp\arduino_build_863737\sketch\macro_test.ino.cpp" -o "nul"
"E:\dev\arduino\arduino-1.8.2\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=136 -DARDUINO=10802 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IE:\dev\arduino\arduino-1.8.2\hardware\teensy\avr\cores\teensy3" "C:\Users\wood\AppData\Local\Temp\arduino_build_863737\sketch\BadLibrary.cpp" -o "nul"
Generating function prototypes...
"E:\dev\arduino\arduino-1.8.2\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=136 -DARDUINO=10802 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IE:\dev\arduino\arduino-1.8.2\hardware\teensy\avr\cores\teensy3" "C:\Users\wood\AppData\Local\Temp\arduino_build_863737\sketch\macro_test.ino.cpp" -o "C:\Users\wood\AppData\Local\Temp\arduino_build_863737\preproc\ctags_target_for_gcc_minus_e.cpp"
"E:\dev\arduino\arduino-1.8.2\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\wood\AppData\Local\Temp\arduino_build_863737\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"E:\dev\arduino\arduino-1.8.2\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -c -O1 -g -Wall -ffunction-sections -fdata-sections -nostdlib -MMD -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -D__MK66FX1M0__ -DTEENSYDUINO=136 -DARDUINO=10802 -DF_CPU=180000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IE:\dev\arduino\arduino-1.8.2\hardware\teensy\avr\cores\teensy3" "C:\Users\wood\AppData\Local\Temp\arduino_build_863737\sketch\BadLibrary.cpp" -o "C:\Users\wood\AppData\Local\Temp\arduino_build_863737\sketch\BadLibrary.cpp.o"
In file included from C:\Users\wood\AppData\Local\Temp\arduino_build_863737\sketch\BadLibrary.cpp:1:0:

BadLibrary.h:3: error: 'typdef' does not name a type
 typdef struct some_struct {

 ^

BadLibrary.h:5: error: 'SOME_STRUCT' does not name a type
 } SOME_STRUCT;

   ^

'typdef' does not name a type
```

Robot-Will, 2017-03-23:
I saw you are trying to build sketches on RPI, this plugin will help. My another repository pyarduino (https://github.com/Robot-Will/pyarduino) will be a common python tool to build sketches after this plugin done. On RPI there is no SublimeText, so this plugin can not work on RPI. But this plugin does not need ST and can run on any OS with python.

KurtE, 2017-03-23:
Hi again, not sure what is different I also just tried with 1.8.2 Beta4 and it compiled on Arduino... 

Again making sure we are on the same page on where you are storing the three files?   
That is My three files are stored at: 
    C:\Users\kurte\Documents\Arduino\BadLIbraryTest  (the .ino file is in this folder) 
    C:\Users\kurte\Documents\Arduino\libraries\BadLibrary (the .h and .cpp file are in this folder)

i.e. - The sketch is in my sketch folder and the library is in the sketch folder/libraries folder... 

If all three files are in sketch folder I believe Arduino will always try to build the .cpp file. 

Thanks for the heads up on your other plugin.  May try it out later.  

So far it looks like the arduino CLI stuff will work for this when I need it.  You can install Arduino and Teensyduino on RPI or Odroid, but it requires gui....  So been playing around with being able to build over CLI and have it working.  Although still learning the CLI options... But I can have a putty window open, plus use WinSCP to transfer files over to RPI and/or double click on file, which brings a temporary version over to PC that I have edit in sublime text and when I do saves WinSCP copies the updated file back.... 

Alternatively: can build the hex file on PC and transfer the hex file to RPI and then use the teensy_loader_cli to program the Teensy...   Who knows maybe at some point setup a way to do this way some place directnly in SublimeText and maybe with help from Stino...

Robot-Will, 2017-03-23:
Good idea. Build sketch on PC and send the binary file to RPI, then upload it to a board. I will think about it. Maybe it is feasible. RPI is not as powerful as PC, build on PC will save a lot of time.

Robot-Will, 2017-03-24:
Maybe because this command.

```
Detecting libraries used...
"D:\arduino-1.8.1\hardware\teensy/../tools/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=136 -DARDUINO=10801 -DF_CPU=96000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-ID:\arduino-1.8.1\hardware\teensy\avr\cores\teensy3" "C:\Users\kurte\AppData\Local\Temp\arduino_build_228645\sketch\BadLIbraryTest.ino.cpp" -o "nul"
```

Robot-Will, 2017-03-24:
Fixed. Thanks.

closed 2017-03-24 by Robot-Will