Robot-Will/Stino #388: Single quotes in side a comment block break the syntax highlighting.

fake-name, 2017-03-19:
[image]

Also, it appears block comments in general are mis-highlighted, albeit not as badly (it doesn't break following code highlighting:

[image]

fake-name, 2017-03-19:
It looks like there's something going on with the highlighting at the start of the file in general:

[image]

Robot-Will, 2017-03-20:
Thanks. This problem is always here and I have no idea how to do with it. Arduino.sublime-syntax is the syntax file and it is simple:

```

%YAML 1.2
---
# http://www.sublimetext.com/docs/3/syntax.html
name: Arduino
file_extensions:
  - ino
  - pde
scope: source.arduino

contexts:
  main:
    - match: .
      push: Packages/C++/C++.sublime-syntax
      with_prototype:
        - match: \b(HIGH|LOW|INPUT|OUTPUT|INPUT_PULLUP|LED_BUILTIN)\b
          scope: constant.language.arduino
        - match: \b(boolean|word|String|string|array)\b
          scope: storage.type.arduino
        - match: PROGRAM
          scope: storage.modifier.arduino
        - match: \b(Serial|Stream|Keyboard|Mouse)\b
          scope: entity.name.class.arduino
        - match: \b(pinMode|digitalWrite|digitalRead|analogReference|analogRead|analogWrite|analogReadResolution|analogWriteResolution|tone|noTone|shiftOut|shiftIn|pulseIn|millis|micros|delay|delayMicroseconds|min|max|constrain|map|pow|sqrt|sin|cos|tan|isAlphaNumeric|isAlpha|isAscii|isWhitespace|isControl|isDigit|isGraph|isLowerCase|isPrintable|isPunct|isSpace|isUpperCase|isHexadecimalDigit|randomSeed|random|lowByte|highByte|bitRead|bitWrite|bitSet|bitClear|bit|attachInterrupt|detachInterrupt|interrupts|noInterrupts)\b
          scope: entity.name.function.arduino
```

Maybe someone would imporve this file and the highlighting will be right.

closed 2017-03-23 by Robot-Will