Robot-Will/Stino #391: arduino-1.6.12  STM32F103C not Build

beho1der, 2017-03-24:
I use arduino-1.6.12 and https://github.com/rogerclarkmelbourne/Arduino_STM32  Arduino SAM Boards. On Arduino IDE all works. Install Stino and Set up:
1.Install Platform --> Add Arduino IDE --> select patch
2. Platform --> Arduino IDE --> Arduino_STM32
3. Version --> STM32F1
4. Board --> Generic STM32F103C series
5. Board option --> Set up for my plate

When I use BUILD not Nothing happens
Image: http://www.picshare.ru/view/7992100/

Robot-Will, 2017-03-24:
Please show me the sketch and library source files. Thanks. I will test the sketch.

beho1der, 2017-03-24:
That is simlpe sketch http://pastebin.com/Rg8x2tF9 and libraries http://dropmefiles.com/YTkys

Robot-Will, 2017-03-28:
I added codes to find build tools before build, you can try it. Thanks.

Robot-Will, 2017-03-28:
The sketch I passed build.

```
[Build] D:/Documents/Arduino/ultra...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[1.4%] Compiling ultra.ino.cpp...
[2.7%] Compiling Ultrasonic.cpp...
[4.1%] Compiling wiring_pulse_f1.cpp...
[5.5%] Compiling wirish_debug.cpp...
[6.8%] Compiling wirish_digital_f1.cpp...
[8.2%] Compiling cxxabi-compat.cpp...
[9.6%] Compiling ext_interrupts.cpp...
[11.0%] Compiling HardwareSerial.cpp...
[12.3%] Compiling HardwareTimer.cpp...
[13.7%] Compiling IPAddress.cpp...
[15.1%] Compiling main.cpp...
[16.4%] Compiling new.cpp...
[17.8%] Compiling Print.cpp...
[19.2%] Compiling pwm.cpp...
[20.5%] Compiling Stream.cpp...
[21.9%] Compiling tone.cpp...
[23.3%] Compiling usb_serial.cpp...
[24.7%] Compiling wirish_analog.cpp...
[26.0%] Compiling wirish_digital.cpp...
[27.4%] Compiling wirish_math.cpp...
[28.8%] Compiling wirish_shift.cpp...
[30.1%] Compiling wirish_time.cpp...
[31.5%] Compiling WString.cpp...
[32.9%] Compiling dtostrf.c...
[34.2%] Compiling usb.c...
[35.6%] Compiling usb_cdcacm.c...
[37.0%] Compiling usb_reg_map.c...
[38.4%] Compiling usb_core.c...
[39.7%] Compiling usb_init.c...
[41.1%] Compiling usb_mem.c...
[42.5%] Compiling usb_regs.c...
[43.8%] Compiling adc.c...
[45.2%] Compiling adc_f1.c...
[46.6%] Compiling bkp_f1.c...
[47.9%] Compiling dac.c...
[49.3%] Compiling dma.c...
[50.7%] Compiling dma_f1.c...
[52.1%] Compiling exti.c...
[53.4%] Compiling exti_f1.c...
[54.8%] Compiling flash.c...
[56.2%] Compiling fsmc_f1.c...
[57.5%] Compiling gpio.c...
[58.9%] Compiling gpio_f1.c...
[60.3%] Compiling i2c.c...
[61.6%] Compiling i2c_f1.c...
[63.0%] Compiling iwdg.c...
[64.4%] Compiling nvic.c...
[65.8%] Compiling pwr.c...
[67.1%] Compiling rcc.c...
[68.5%] Compiling rcc_f1.c...
[69.9%] Compiling spi.c...
[71.2%] Compiling spi_f1.c...
[72.6%] Compiling systick.c...
[74.0%] Compiling timer.c...
[75.3%] Compiling timer_f1.c...
[76.7%] Compiling usart.c...
[78.1%] Compiling usart_f1.c...
[79.5%] Compiling usart_private.c...
[80.8%] Compiling util.c...
[82.2%] Compiling util_hooks.c...
[83.6%] Compiling hooks.c...
[84.9%] Compiling itoa.c...
[86.3%] Compiling isrs.S...
[87.7%] Compiling vector_table.S...
[89.0%] Compiling exc.S...
[90.4%] Compiling boards.cpp...
[91.8%] Compiling boards_setup.cpp...
[93.2%] Compiling board.cpp...
[94.5%] Compiling start_c.c...
[95.9%] Compiling syscalls.c...
[97.3%] Compiling start.S...
[98.6%] Linking everything together...
[100.0%] Creating binary files...
Sketch uses 15,940 bytes (24.3%) of program storage space. Maximum is 65,536 bytes.
Global variables use 3,120 bytes (31.2%) of dynamic memory, leaving 6,880 bytes for local variables. Maximum is 10,000 bytes.
Build done.
```

beho1der, 2017-03-28:
Big thanks, now everything works!

closed 2017-03-28 by beho1der