Robot-Will/Stino #392: Core.a not found

grant-rez, 2017-03-25:
Hello,

I'm trying to get this plug-in to work, but I can't seem to get it done. I have tried a clean install of sublime, but that hasn't helped. This is the error I get specifically:
`avr-gcc: error: C:/Users/grant/AppData/Local/Arduino15/build/SunTracker/core.a: No such file or directory`
I have tried compiling my code with examples as well as the actual code I would like to compile, but none of it will compile. 

Do I need to have the Arduino I intend to use attached to my computer when I compile? This is the only thing that I could possibly be doing wrong. 

Also, I'm not sure if it's a known issue, but whenever I try to highlight too more than about 100 characters or so in my Stino console, Sublime freezes.

Robot-Will, 2017-03-25:
Yes, the console freezes, and I do a lot for this problem. The upload process I improved and you can try the DEV branch and see the result. Thanks.

Nume1977, 2017-03-25:
I have the same bug. I installed the DEV Branch and it works, but there is an error on DEV Branch

Robot-Will, 2017-03-27:
@grant-rez  I need a test sketch, thanks.

grant-rez, 2017-03-28:
The new dev branch fixed this problem it seems.

Thanks

grant-rez, 2017-03-28:
Sorry, but I'm getting an error:
`In file included from C:/Program Files (x86)/Arduino/libraries/Servo/examples/Knob/Knob.ino:10:0:
C:/Program Files (x86)/Arduino/libraries/Servo/src/Servo.h:73:2: error: #error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."
 #error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."
  ^
[Build] Error occurred.`

I'm using the Knob example file from Servo and Arduino AVR Boards, version 1.6.17, and Arduino/Genuino Uno options.

Robot-Will, 2017-03-30:
@grant-rez Fixed. Thanks.

closed 2017-03-30 by Robot-Will