Robot-Will/Stino #393: Dev Branch #include bug

Nume1977, 2017-03-25:
The "avr-gcc: error: C:/Users/xxxxx/AppData/Local/Arduino15/build/yourcode/core.a: No such file or directory" is solved in DEV Branch, but now there is a new bug present.

If we import a library  it will fail to compile at the include, example:
`#include <PCF8574.h>` This will fail saying "PCF8574.h : No such file or directory".

But if i use the full path to the library it will work, example:
`#include <C:\User\me\Arduino\Libraries\PCF8574.h>`

Hope this helps!
Keep up the good work!

Robot-Will, 2017-03-25:
Is the lib in sketchbook? Show me a test case and I will find the reason. Thanks.

paulcarbone, 2017-03-26:
I'm getting what may be a similar issue, both on Mac & Linux, on both stable & dev branch.   

When I point my sketchbook path to an existing folder (~/development/arduino), 'import library' menu shows the installed libraries, but I still have to use the full path to compile.  

When using either the default sketchbook path, or my existing one, I cannot install libraries.  When I use my existing path, and try to install a library that already exists, I do get the warning dialog box. 
In the console, the error I get when trying to install a library: 
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/paul/Library/Application Support/Sublime Text 3/Packages/Stino-Dev/libs/base_utils/task_queue.py", line 88, in _run
    self._consumer(*args)
TypeError: download_lib() takes 1 positional argument but 79 were given

I will get this error only once per session.  All other attempts fail silently. 

Let me know what else I can do to help.  I'm really excited to use this, and thanks for all your work!

Robot-Will, 2017-03-27:
Master branch works, I am working to find out what's wrong with the new build process in dev branch. Thanks.

Robot-Will, 2017-03-27:
@paulcarbone Install library I fixed. Thanks.

Robot-Will, 2017-03-27:
Now Dev branch works. Thanks.

paulcarbone, 2017-03-27:
Works great in both OSX & Linux, thanks!!

Nume1977, 2017-03-27:
Just downloaded the latest Stable and Dev, and they both fail with the "avr-gcc: core.a: No such file or directory" error now...

Robot-Will, 2017-03-28:
@Nume1977 Which platform, show me the link? Install from package index file or install into arduino IDE?

Robot-Will, 2017-03-28:
The following core works.

Atmega & Attiny cores: http://www.leonardomiliani.com/repository/package_leonardomiliani.com_index.json

ATmega644/644P & ATmega1284P core (w/bootloader)
ATmega168P/328P core (for standalone MCUs)
ATtiny24/44/84, ATtiny25/45/85, and ATtiny2313/4313 core (I2C & SoftSerial)

closed 2017-03-30 by Robot-Will