Robot-Will/Stino #394: Expressif ESP8266 tools fail to compile

fake-name, 2017-03-26:
Platform is Arduino 1.8.1, with both the ESP32 and ESP8266 tools installed from git.

The normal arduino boards compile fine, as do any ESP32 projects. However, when I try to build a ESP8266 project, it fails.

ESP8266 tooling is version 2.3.0 installed via `http://arduino.esp8266.com/stable/package_esp8266com_index.json`.

The build output stalls at: 

```
[Build] D:/Profile/Desktop/Repository/OpenSource-old/WeatherStation/EspLogger - R1...
[Step 1] Check Toolchain.
[Step 2] Find all source files.

```

Opening the console, I see the following:

```

Could not import subprocess32 module, falling back to subprocess module
Exception in thread Thread-13:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "C:\Users\durr\AppData\Roaming\Sublime Text 3\Packages\Stino\libs\base_utils\task_queue.py", line 88, in _run
    self._consumer(*args)
  File "C:\Users\durr\AppData\Roaming\Sublime Text 3\Packages\Stino\libs\stino_runtime\__init__.py", line 1640, in build_sketch
    cmd_preproc_includes = cmds_info['recipe.preproc.includes']
KeyError: 'recipe.preproc.includes'

Package Control: No updated packages

```

Adding some debugging to the `__init__.py` the contents of `cmds_info` are the following:

```
 
{
	'recipe.S.o.pattern': '"C:/Users/durr/AppData/Local/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include" "-IC:/Users/durr/AppData/Local/Arduino15/build/Esp8266Logger/core" -c -g -x assembler-with-cpp -MMD -mlongcalls -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=20000 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_esp8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:/Users/durr/AppData/Local/Arduino15/build/Esp8266Logger" "-ID:/Profile/Desktop/arduino-1.8.1/libraries/WiFi/src" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/include" "-ID:/Profile/Desktop/arduino-1.8.1/libraries/WiFi/src/utility" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/variants/generic" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/spiffs" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/libb64" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/SPI" "-ID:/Profile/Desktop/Repository/OpenSource-old/WeatherStation/Esp8266Logger" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/lib/gcc/xtensa-lx106-elf/4.8.2/include" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/lib/gcc/xtensa-lx106-elf/4.8.2/install-tools/include" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include" "{source_file}" -o "{object_file}"',
	'recipe.ar.pattern': '"C:/Users/durr/AppData/Local/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru  "C:/Users/durr/AppData/Local/Arduino15/build/Esp8266Logger/arduino.ar" "{object_file}"',
	'recipe.c.combine.pattern': '"C:/Users/durr/AppData/Local/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -g -w -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static "-LC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lib" "-LC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/ld" "-Teagle.flash.4m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy  -o "C:/Users/durr/AppData/Local/Arduino15/build/Esp8266Logger/Esp8266Logger.elf" -Wl,--start-group {object_files} "C:/Users/durr/AppData/Local/Arduino15/build/Esp8266Logger/arduino.ar" -lm -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lcrypto -lmain -lwps -laxtls -lsmartconfig -lmesh -lwpa2 -llwip_gcc -lstdc++ -Wl,--end-group  "-LC:/Users/durr/AppData/Local/Arduino15/build/Esp8266Logger"',
	'recipe.c.o.pattern': '"C:/Users/durr/AppData/Local/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include" "-IC:/Users/durr/AppData/Local/Arduino15/build/Esp8266Logger/core" -c -w -Os -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -std=gnu99 -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=20000 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_esp8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:/Users/durr/AppData/Local/Arduino15/build/Esp8266Logger" "-ID:/Profile/Desktop/arduino-1.8.1/libraries/WiFi/src" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/include" "-ID:/Profile/Desktop/arduino-1.8.1/libraries/WiFi/src/utility" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/variants/generic" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/spiffs" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/libb64" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/SPI" "-ID:/Profile/Desktop/Repository/OpenSource-old/WeatherStation/Esp8266Logger" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/lib/gcc/xtensa-lx106-elf/4.8.2/include" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/lib/gcc/xtensa-lx106-elf/4.8.2/install-tools/include" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include" "{source_file}" -o "{object_file}"',
	'recipe.cpp.o.pattern': '"C:/Users/durr/AppData/Local/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include" "-IC:/Users/durr/AppData/Local/Arduino15/build/Esp8266Logger/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=20000 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_esp8266 -DARDUINO_BOARD="ESP8266_ESP01"  -DESP8266 "-IC:/Users/durr/AppData/Local/Arduino15/build/Esp8266Logger" "-ID:/Profile/Desktop/arduino-1.8.1/libraries/WiFi/src" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src/include" "-ID:/Profile/Desktop/arduino-1.8.1/libraries/WiFi/src/utility" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/variants/generic" "-IC:/Users/durr/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/spiffs" "-IC:/Users/durr/AppData/Local/Arduino15/packages/ [... cut ...] k/Blink.elf" -Wl,--start-group "C:/Users/durr/AppData/Local/Arduino15/build/Blink/Blink.ino.cpp.o" "C:/Users/durr/AppData/Local/Arduino15/build/Blink/arduino.ar" -lm -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lcrypto -lmain -lwps -laxtls -lsmartconfig -lmesh -lwpa2 -llwip_gcc -lstdc++ -Wl,--end-group  "-LC:/Users/durr/AppData/Local/Arduino15/build/Blink"
xtensa-lx106-elf-gcc: error: C:/Users/durr/AppData/Local/Arduino15/build/Blink/arduino.ar: No such file or directory
[Build] Error occurred.

```

fake-name, 2017-03-26:
Looking at the contents of `cmds_info`, I think maybe the `'recipe.ar.pattern'` command creates the archive file?

Robot-Will, 2017-03-27:
Master branch works, I am working to find out what's wrong with the new build process in dev branch. Thanks.

fake-name, 2017-03-27:
Oh, I guess I'm only looking at the dev branch because I'm using arduino 1.8.1.

Robot-Will, 2017-03-27:
Master branch and Dev branch are all stino2017. The old version I deleted, and now the master branch is a reletive less problem version, and the Dev version is the version under developement. The codes you are trying to modify is in master branch currently, not in Dev branch.

Robot-Will, 2017-03-27:
Now Dev branch works. Thanks.

fake-name, 2017-03-27:
Awesome!  I'll check it out tonight!

closed 2017-03-30 by Robot-Will