Robot-Will/Stino #396: WinError 5] : Access denied

Alvariux510, 2017-03-26:
Hello, i have a problem when a try to build a proyect, anyone.

i have Installed IDE 1.8.2  and works correctly, I can compile projects from proteus, without any problem
[image]

but I can not compile from stino, The message: Check Toolchain appears on the panel, And on the console
WinError 5] Acceso denegado: 'C:\\Users\\Alvariux\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.9.2-atmel3.5.3-arduino2'

what am I doing wrong?

Robot-Will, 2017-03-27:
I found a link, maybe help:
http://stackoverflow.com/questions/18541458/python-windowserror-5-access-denied

Alvariux510, 2018-11-21:
> Hello, i have a problem when a try to build a proyect, anyone.
> 
> i have Installed IDE 1.8.2 and works correctly, I can compile projects from proteus, without any problem
> [image]
> 
> but I can not compile from stino, The message: Check Toolchain appears on the panel, And on the console
> WinError 5] Acceso denegado: 'C:\Users\Alvariux\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2'
> 
> what am I doing wrong? thanks for your help

closed 2018-11-21 by Alvariux510