Robot-Will/Stino #397: How to build?

noumenonquest, 2017-03-27:
After adding the Arduino IDE platform, when i try to build  the following appears.
`[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.`

What should i do in each step?

Robot-Will, 2017-03-27:
It seems error ocurred. 
Please afford the detailed information: the OS, plugin branch, the platform, the sketch and libraries.
And ctrl+`, show me the error messages. Thanks.

Robot-Will, 2017-03-27:
The following messages you will get if it works.

```
[Build] D:/Documents/Arduino/Blink...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[3.6%] Compiling Blink.ino.cpp...
[7.1%] Compiling abi.cpp...
[10.7%] Compiling CDC.cpp...
[14.3%] Compiling HardwareSerial.cpp...
[17.9%] Compiling HardwareSerial0.cpp...
[21.4%] Compiling HardwareSerial1.cpp...
[25.0%] Compiling HardwareSerial2.cpp...
[28.6%] Compiling HardwareSerial3.cpp...
[32.1%] Compiling IPAddress.cpp...
[35.7%] Compiling main.cpp...
[39.3%] Compiling new.cpp...
[42.9%] Compiling PluggableUSB.cpp...
[46.4%] Compiling Print.cpp...
[50.0%] Compiling Stream.cpp...
[53.6%] Compiling Tone.cpp...
[57.1%] Compiling USBCore.cpp...
[60.7%] Compiling WMath.cpp...
[64.3%] Compiling WString.cpp...
[67.9%] Compiling hooks.c...
[71.4%] Compiling WInterrupts.c...
[75.0%] Compiling wiring.c...
[78.6%] Compiling wiring_analog.c...
[82.1%] Compiling wiring_digital.c...
[85.7%] Compiling wiring_pulse.c...
[89.3%] Compiling wiring_shift.c...
[92.9%] Compiling wiring_pulse.S...
[96.4%] Linking everything together...
[100.0%] Creating binary files...
Sketch uses 4,206 bytes (14.7%) of program storage space. Maximum is 28,672 bytes.
Global variables use 163 bytes (6.4%) of dynamic memory, leaving 2,397 bytes for local variables. Maximum is 2,560 bytes.
```

noumenonquest, 2017-03-27:
Thanks for the reply, i found out what caused it to fail to compile, it seems as though it is necessary to add the void loop() section in the code in order to compile it successfully.

closed 2017-03-27 by noumenonquest