Robot-Will/Stino #400: Compiling without setup() and loop() functions.

noumenonquest, 2017-03-29:
The arduino ide allows compiling and uploading avr programs written in C/C++ language, with the code written in a main() function. (without the setup() and loop() functions), However, the avr program does not  compile in sublime(stino), do i need to add some libraries, or avr-gcc?

Robot-Will, 2017-03-30:
ATMEGA2560

The source code:

```
#include <avr/io.h>
#include <util/delay.h>

int main()
{
    DDRC = 0xFF;
    while(1)
    {
        PORTC ^= _BV(PC0);
        _delay_ms(1000);
    }
}
```

The build output:

```
[Build] D:/Documents/Arduino/avr01...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[50.0%] Compiling avr01.ino.cpp...
"C:/Users/wood/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=20000 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-ID:/Documents/Arduino/avr01" "-IC:/Users/wood/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/avr/include" "-IC:/Users/wood/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/i686-pc-mingw32/avr/include" "-IC:/Users/wood/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/include" "-IC:/Users/wood/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/lib/gcc/avr/4.9.2/include" "-IC:/Users/wood/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/lib/gcc/avr/4.9.2/install-tools/include" "-IC:/Users/wood/AppData/Local/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino8/include" "C:/Users/wood/AppData/Local/Arduino15/build/avr01/sketch/avr01.ino.cpp" -o "C:/Users/wood/AppData/Local/Arduino15/build/avr01/sketch/avr01.ino.cpp.o"
[100.0%] Creating binary files...
"C:/Users/wood/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega2560  -o "C:/Users/wood/AppData/Local/Arduino15/build/avr01/avr01.elf" "C:/Users/wood/AppData/Local/Arduino15/build/avr01/sketch/avr01.ino.cpp.o"  "-LC:/Users/wood/AppData/Local/Arduino15/build/avr01" -lm
"C:/Users/wood/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0  "C:/Users/wood/AppData/Local/Arduino15/build/avr01/avr01.elf" "C:/Users/wood/AppData/Local/Arduino15/build/avr01/avr01.eep"
"C:/Users/wood/AppData/Local/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-objcopy" -O ihex -R .eeprom  "C:/Users/wood/AppData/Local/Arduino15/build/avr01/avr01.elf" "C:/Users/wood/AppData/Local/Arduino15/build/avr01/avr01.hex"
Sketch uses 292 bytes (0.1%) of program storage space. Maximum is 253,952 bytes.
Global variables use 0 bytes (0.0%) of dynamic memory, leaving 8,192 bytes for local variables. Maximum is 8,192 bytes.
Build done.
```

Thanks.

closed 2017-03-30 by Robot-Will

noumenonquest, 2017-03-30:
Thanks, works for .c, .cpp, and .ino extensions!