Robot-Will/Stino #404: Wishlist: Ability to upload image to remote processor

KurtE, 2017-03-30:
Sorry, not sure if this is the best place for possible feature requests.  

There are times when I have a Teensy 3.x attached to a Raspberry Pi or the like (Odroid, UP...)
I currently have it setup that I can do all of it on the RPI3, including now even only needing to use an SSH window like PuTTy.  But it is not always the nicest way to work. 

It would be great if I could do all of the development on my main machine using SublimeText and Stino.  
It would be great if I could somehow do the build using Stino and have the binary (hex) file transferred to the remote board (something like PSCP) and potentially invoke a command on the remote machine to program the Teensy (or other Arduino).   I sort of used to do this with Edison boards with some of the stuff built into Eclipse.

As I mentioned on the PJRC forum: https://forum.pjrc.com/threads/42790-Arduino-builds-using-Command-lines-for-teensy-using-PuTTy-or-the-like?p=137917&viewfull=1#post137917

I am able to do some of this, in a clunky manner.   
Example If I use Arduino IDE command "Export Compiled Binary", it builds the sketch and puts the hex file into the sketch folder (Might be nice to add that to Stino)

Then I had a WinSCP folder open to the sketch folder on both boards and after the compile was done, I dragged and dropped the hex file to copy it over to the RPI3.

I then had a task running on the RPI3, that programmed the Teensy.
```
#!/bin/sh

while :
do
	inotifywait Teensy_Arbotix_Pro.ino.TEENSY31.hex
	~/teensy_loader_cli/teensy_loader_cli --mcu=TEENSY31 -v -w -s Teensy_Arbotix_Pro.ino.TEENSY31.hex 
done
```

This is working, but would be nice to do more of this inside Stino.  

Again sorry if this is not an appropriate request.

Thanks for all of your great work on this new version of Stino!

Robot-Will, 2017-03-31:
Sure, this is my plan. But before do this, I will learn something about SSH. Thanks.

Robot-Will, 2017-03-31:
Now, after each build a binary file is copied to the sketch folder. Then I will learn SSH.