Robot-Will/Stino #408: Auto Reformat lost Switching board sketches between boards

Defragster, 2017-03-31:
I have Teensy and STM32 sketches open.

Auto Format is lost in one or the other after UPLOAD from the other.

Open Sublime Teensy sketch, do Auto Format, upload to Teensy, go to STM32 sketch and it won't do Auto Format.

Error Text may be this?

> Traceback (most recent call last):
  File "C:\Program Files\Sublime Text 3\sublime_plugin.py", line 818, in run_
    return self.run(edit)
  File "C:\Users\Tim\AppData\Roaming\Sublime Text 3\Packages\Stino-Stino2017\StinoCommands.py", line 1306, in run
    stino.beautify_src(self.view, edit, file_path)
  File "C:\Users\Tim\AppData\Roaming\Sublime Text 3\Packages\Stino-Stino2017\libs\stino_runtime\__init__.py", line 2257, in beautify_src
    beautiful_text = cur_file.get_beautified_text()
  File "C:\Users\Tim\AppData\Roaming\Sublime Text 3\Packages\Stino-Stino2017\libs\base_utils\c_file.py", line 1057, in get_beautified_text
    self._beautified_lines = beautify_lines(self._lines)
  File "C:\Users\Tim\AppData\Roaming\Sublime Text 3\Packages\Stino-Stino2017\libs\base_utils\c_file.py", line 922, in beautify_lines
    lines = indent_lines(lines)
  File "C:\Users\Tim\AppData\Roaming\Sublime Text 3\Packages\Stino-Stino2017\libs\base_utils\c_file.py", line 634, in indent_lines
    last_slice = line_slices[-2]
IndexError: list index out of range

Robot-Will, 2017-03-31:
I need the source file to find the reason. Thanks.

Defragster, 2017-03-31:
Attached are the INO sketch files for the Teensy T_3.6 and the STM32 LadyBug I was using
[adcdma_stm32.ino.txt](https://github.com/Robot-Will/Stino/files/886398/adcdma_stm32.ino.txt)
[Acorr_Teensy.ino.txt](https://github.com/Robot-Will/Stino/files/886402/Acorr_Teensy.ino.txt)

Robot-Will, 2017-03-31:
I fixed a few codes. Thanks.

Defragster, 2017-03-31:
I pulled that change and the Auto Format worked - but it introduced CODE ERRORS.

I started with the INO posted above and attached is what I got after a Reformat.

[adcdma_BAD_FORMAT.ino.txt](https://github.com/Robot-Will/Stino/files/886605/adcdma_BAD_FORMAT.ino.txt)

Here are the error lines pointed out:
`d:/Users/Tim/Documents/GitHub/STM32L4_tests/adcdma/adcdma.ino:4:2: error: invalid preprocessing directive #x
 #x " 0x");
  ^
d:/Users/Tim/Documents/GitHub/STM32L4_tests/adcdma/adcdma.ino:19:53: error: "|" may not appear in macro parameter list
 #define DMA_OPTIONS(DMA_OPTION_PERIPHERAL_TO_MEMORY | DMA_OPTION_MEMORY_DATA_SIZE_16 | DMA_OPTION_PERIPHERAL_DATA_SIZE_16 | DMA_OPTION_MEMORY_DATA_INCREMENT | DMA_OPTION_MEMORY_DATA_INCREMENT | DMA_OPTION_EVENT_TRANSFER_DONE)
                                                     ^
d:/Users/Tim/Documents/GitHub/STM32L4_tests/adcdma/adcdma.ino:5:1: error: 'Serial' does not name a type
 Serial.println(x, HEX) // Debug print HEX
 ^
d:/Users/Tim/Documents/GitHub/STM32L4_tests/adcdma/adcdma.ino: In function 'void timer_isr(void*, uint32_t)':
d:/Users/Tim/Documents/GitHub/STM32L4_tests/adcdma/adcdma.ino:15:3: error: 'ticks' was not declared in this scope
   ticks++;
   ^
d:/Users/Tim/Documents/GitHub/STM32L4_tests/adcdma/adcdma.ino: In function 'void readADC(bool)':
d:/Users/Tim/Documents/GitHub/STM32L4_tests/adcdma/adcdma.ino:80:87: error: 'DMA_OPTIONS' was not declared in this scope
   stm32l4_dma_start(& dma,(uint32_t) adcbuf[0],(uint32_t) & (ADC1 -> DR), ADCBUF_CNT, DMA_OPTIONS);
                                                                                       ^
d:/Users/Tim/Documents/GitHub/STM32L4_tests/adcdma/adcdma.ino: In function 'void stm32l4_ADC_dma_callback(void*, uint32_t)':
d:/Users/Tim/Documents/GitHub/STM32L4_tests/adcdma/adcdma.ino:92:87: error: 'DMA_OPTIONS' was not declared in this scope
   stm32l4_dma_start(& dma,(uint32_t) adcbuf[0],(uint32_t) & (ADC1 -> DR), ADCBUF_CNT, DMA_OPTIONS);
                                                                                       ^
d:/Users/Tim/Documents/GitHub/STM32L4_tests/adcdma/adcdma.ino: In function 'void setup()':
d:/Users/Tim/Documents/GitHub/STM32L4_tests/adcdma/adcdma.ino:112:87: error: 'DMA_OPTIONS' was not declared in this scope
   stm32l4_dma_start(& dma,(uint32_t) adcbuf[0],(uint32_t) & (ADC1 -> DR), ADCBUF_CNT, DMA_OPTIONS);
                                                                                       ^
d:/Users/Tim/Documents/GitHub/STM32L4_tests/adcdma/adcdma.ino: In function 'void loop()':
d:/Users/Tim/Documents/GitHub/STM32L4_tests/adcdma/adcdma.ino:185:18: error: 'ticks' was not declared in this scope
     Serial.print(ticks);
                  ^
[Build] Error occurred.
`

Defragster, 2017-04-01:
I can switch between board sketches and Auto Format and Upload without issue.

It compiles without error and runs, but Auto Format is still doing something odd with these two lines:
`#define PRREG(x) Serial.print(#x" 0x"); Serial.println(x,HEX) // Debug print HEX
#define qBlinkg() (digitalWrite(A3, !digitalRead(A3) ))`

becomes three lines and extra space before the #define where the prior lined ends without a semi-colon:
`#define PRREG(x) Serial.print(#x " 0x");
Serial.println(x, HEX) // Debug print HEX
  #define qBlinkg() (digitalWrite(A3, !digitalRead(A3)))
`

Defragster, 2017-04-01:
Looks good - fixed that #define line break

closed 2017-04-01 by Defragster