Robot-Will/Stino #411: Compilation error

olivernadj, 2017-04-02:
I have a similar problem like in  #228 but when I try to compile the architecture definition has a lowercase processor suffix. `-DARDUINO_ARCH_avr`

Here is the build:

```
[Build] /home/oliver/work/R3/sketchbook/servo...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[3.4%] Compiling servo.ino.cpp...
"/home/oliver/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=20000 -DARDUINO_AVR_UNO -DARDUINO_ARCH_avr   "-I/home/oliver/work/R3/sketchbook/servo" "-I/home/oliver/.arduino15/packages/arduino/hardware/avr/1.6.17/cores/arduino" "-I/home/oliver/.arduino15/packages/arduino/hardware/avr/1.6.17/variants/standard" "-I/home/oliver/work/R3/sketchbook/libraries/Servo/src" "-I/home/oliver/work/R3/sketchbook/libraries/Servo/src/avr" "/home/oliver/.arduino15/build/servo/sketch/servo.ino.cpp" -o "/home/oliver/.arduino15/build/servo/sketch/servo.ino.cpp.o"
In file included from /home/oliver/work/R3/sketchbook/servo/servo.ino:4:0:
/home/oliver/work/R3/sketchbook/libraries/Servo/src/Servo.h:73:2: error: #error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."
 #error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."
  ^
[Build] Error occurred.
```
The Stino version is: `v2017.04.02.03.05.06` installed by Package manager.

Robot-Will, 2017-04-02:
Fixed. Thanks.

olivernadj, 2017-04-03:
Thanks S. Zhang,

It works now.

closed 2017-04-03 by olivernadj