Robot-Will/Stino #412: Compile issue: functions that take a reference to class variable?

KurtE, 2017-04-02:
I am on Windows 10, Arduino 1.8.2 and Teensyduino 1.36 (just released), have Teensy 3.2 selected as board type.  I am up to sync with your current stuff.  

I found yesterday with a test program, that when I changed a function from having no parameters, to one where I passed a reference to the display object (experimenting supporting multiple displays in library), that the compile using Stino failed... It compiles fine using Arduino...  This morning I tried doing it again with a different library (one installed with Teensyduino) and was able to replicate the issue. 
```
#include <font_Arial.h>
#include <font_ArialBold.h>
#include <ILI9341_t3.h>

/******************************************************************************
   Original Creation Date: October 27, 2014
   Modified Febuary 2, 2017

   This code is beerware; if you see me (or any other SparkFun employee) at thewa
   local, and you've found our code helpful, please buy us a round!

   Distributed as-is; no warranty is given.
 ******************************************************************************/
#define TFT_DC  9
#define TFT_CS 10
#define TFT_RST 7
#define TFT_SCK 13
#define TFT_MISO 12
#define TFT_MOSI 11

#define TFT_CS2 20
#define TFT_RST2 6

ILI9341_t3 tft = ILI9341_t3(TFT_CS, TFT_DC, TFT_RST, TFT_MOSI, TFT_SCK, TFT_MISO);
ILI9341_t3 tft2 = ILI9341_t3(TFT_CS2, TFT_DC, TFT_RST2, TFT_MOSI, TFT_SCK, TFT_MISO);

void setup()
{
	while(!Serial && millis() < 3000) ;
	Serial.begin(38400);

  tft.begin();
  tft.setRotation(3);
  tft.fillScreen(ILI9341_BLACK);

  tft2.begin();
  tft2.setRotation(3);
  tft2.fillScreen(ILI9341_BLACK);
}

void FillTest(ILI9341_t3 &disp) {
  disp.drawRect(0, 150, 100, 50, ILI9341_WHITE);
  disp.drawLine(0, 150, 100, 50, ILI9341_GREEN);
  disp.fillRectVGradient(125, 150, 50, 50, ILI9341_GREEN, ILI9341_YELLOW);
  disp.fillRectHGradient(200, 150, 50, 50, ILI9341_YELLOW, ILI9341_GREEN);
}

void loop()
{
	FillTest(tft);
	FillTest(tft2);
	delay(1000);	
}
```
It also fails when I change the function to take a pointer instead of a reference:
```
void FillTest(ILI9341_t3 *disp) {
  disp->drawRect(0, 150, 100, 50, ILI9341_WHITE);
  disp->drawLine(0, 150, 100, 50, ILI9341_GREEN);
  disp->fillRectVGradient(125, 150, 50, 50, ILI9341_GREEN, ILI9341_YELLOW);
  disp->fillRectHGradient(200, 150, 50, 50, ILI9341_YELLOW, ILI9341_GREEN);
}

void loop()
{
	FillTest(&tft);
	FillTest(&tft2);
	delay(1000);	
}
```

Robot-Will, 2017-04-02:
Fixed. Thanks.

KurtE, 2017-04-02:
Thanks I synced up and I am still seeing this issue?  Do I need to do something else to clear it.

Thanks again

Kurt

KurtE, 2017-04-02:
Edit -  
Sort of never mind.  I edited the file and did a save and then it builds.  So probably it had cached the conversion of .ino to .cpp and just used it.

Robot-Will, 2017-04-03:
.ino files cannot be compiled by compiler, and the .ino files are combined to one .cpp file before build. To save time, If no .ino file changed, the  soft do not do the combine and use the last combined .cpp file. Thanks.

closed 2017-04-03 by Robot-Will