Robot-Will/Stino #413: Wish: A way to clear the message panel

KurtE, 2017-04-02:
After I do a compile or two, there are times I would like to be able to clear out the contents of the Arduino message panel.  

Might be nice to have on context menu.  For a minute I thought I could do it using the select all and then use the cut, but it did not actually cut the data... 

Again this is a wish list and not a very high priority item.

Robot-Will, 2017-04-03:
Clear part or all? The panel is readonly, so you can not edit it.

KurtE, 2017-04-03:
Either would be great... 

Sort of like when you do an Arduino compile you no longer see the error messages from the previous build. 
I have at times scrolled back too far... 

Note: on context menu if the area is read only, then the menu item CUT does not make any sense, likiewise paste would not work. should be only be copy...

Robot-Will, 2017-04-03:
I gave a stand-alone build message panel. Thanks.

KurtE, 2017-04-04:
Thanks, 

 

Works great.  One suggestion.  If I do a build and the error messages show up, which is nice, but then I hit the ESC key and the panel is closed, 

I don’t think there is any way to get that panel back.

Robot-Will, 2017-04-04:
You do not want the last build messages back?

In the memu->show build message panel, you can reopen the build panel with the last build messages.

KurtE, 2017-04-06:
Thanks, that works great

closed 2017-04-06 by KurtE