Robot-Will/Stino #415: Arduino Upload runs on C source file

Defragster, 2017-04-04:
I accidentally hit UPLOAD while editing a common system core source file and it compiled and went to upload.

But I have sketches open to 3 different MCU's and I didn't know what it was going to upload so I cancelled.

Would it make sense to limit UPLOAD to *.INO sketch files?  Or will it somehow do the right thing - or compile and upload the last sketch uploaded?

Robot-Will, 2017-04-04:
This plugin is compatibale with none-Arduino projects, such as avr projects not using Arduino core. So it builds C/C++/Arduino source files. 

I added a menu item:
Menu -> Build Options -> Build Enabled

You can disable build/upload manually if you do not need build/upload.

Thanks.

Defragster, 2017-04-04:
Thanks - I had not considered that - no problem as it was.  I'm not sure it is worth yet another menu item.

closed 2017-04-04 by Defragster