Robot-Will/Stino #416: Sketch does not build using Stino...

KurtE, 2017-04-04:
I have a test app that currently has two .ino files (not sure if that is the issue),  It compiles fine on Arduino 1.8.2 with latest Teensyduino 1.36 for Teensy 3.2. 

https://github.com/KurtE/AX12_Test

Gives lots of duplicate defined errors:
```

[Build] C:/Users/kurte/Documents/Arduino/Github/AX12_Test...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[1.8%] Compiling AX12_Test.ino.cpp...
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:85:30: error: redefinition of 'const byte pgm_axdIDs []'
 static const byte pgm_axdIDs[] PROGMEM = {
                              ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:85:19: note: 'const byte pgm_axdIDs [18]' previously defined here
 static const byte pgm_axdIDs[] PROGMEM = {
                   ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:102:25: error: redefinition of 'const char* IKPinsNames []'
 const char* IKPinsNames[] = {
                         ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:102:13: note: 'const char* IKPinsNames [18]' previously defined here
 const char* IKPinsNames[] = {
             ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:123:21: error: redefinition of 'BioloidControllerEx bioloid'
 BioloidControllerEx bioloid = BioloidControllerEx(); 
                     ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:123:21: note: 'BioloidControllerEx bioloid' previously declared here
 BioloidControllerEx bioloid = BioloidControllerEx(); 
                     ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:127:16: error: redefinition of 'word g_wVoltage'
 word           g_wVoltage;
                ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:127:16: note: 'word g_wVoltage' previously declared here
 word           g_wVoltage;
                ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:128:31: error: redefinition of 'char g_aszCmdLine [80]'
 char           g_aszCmdLine[80];
                               ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:128:16: note: 'char g_aszCmdLine [80]' previously declared here
 char           g_aszCmdLine[80];
                ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:129:16: error: redefinition of 'uint8_t g_iszCmdLine'
 uint8_t        g_iszCmdLine;
                ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:129:16: note: 'uint8_t g_iszCmdLine' previously declared here
 uint8_t        g_iszCmdLine;
                ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:130:16: error: redefinition of 'boolean g_fTrackServos'
 boolean        g_fTrackServos = false;
                ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:130:16: note: 'boolean g_fTrackServos' previously defined here
 boolean        g_fTrackServos = false;
                ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:133:15: error: redefinition of 'byte g_bServoID'
 byte          g_bServoID;
               ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:133:15: note: 'byte g_bServoID' previously declared here
 byte          g_bServoID;
               ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:134:15: error: redefinition of 'word g_wServoGoalPos'
 word          g_wServoGoalPos;
               ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:134:15: note: 'word g_wServoGoalPos' previously declared here
 word          g_wServoGoalPos;
               ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:135:15: error: redefinition of 'word g_wServoGoalSpeed'
 word          g_wServoGoalSpeed;
               ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:135:15: note: 'word g_wServoGoalSpeed' previously declared here
[... 121 lines cut ...]
 void TrackPrintMinsMaxs(void) {
      ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:528:6: note: 'void TrackPrintMinsMaxs()' previously defined here
 void TrackPrintMinsMaxs(void) {
      ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino: In function 'void PrintServoValues()':
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:546:6: error: redefinition of 'void PrintServoValues()'
 void PrintServoValues(void) {
      ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:546:6: note: 'void PrintServoValues()' previously defined here
 void PrintServoValues(void) {
      ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino: In function 'boolean GetMultax12Registers(int, int, int, uint8_t*)':
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:578:9: error: redefinition of 'boolean GetMultax12Registers(int, int, int, uint8_t*)'
 boolean GetMultax12Registers(int id, int regstart, int length, uint8_t *pab){  
         ^
C:/Users/kurte/Documents/Arduino/Github/AX12_Test/AX12_Test.ino:578:9: note: 'boolean GetMultax12Registers(int, int, int, uint8_t*)' previously defined here
 boolean GetMultax12Registers(int id, int regstart, int length, uint8_t *pab){  
         ^
[Build] Error occurred.
```

Almost like maybe it included the stuff twice?   

I do have a few changes local to my machine, to this sketch, but they should not matter.   Only real difference is on a couple of pins being defined. Could upload changes if needed

Robot-Will, 2017-04-04:
It's due to a typo error, and I fixed. Still I can not pass the build. I will continue to find re reason. Thanks.

Robot-Will, 2017-04-06:
I fixed. Macro caused the build problems. Thanks.

closed 2017-04-07 by Robot-Will