Robot-Will/Stino #417: Adafruit_ssd1306 example did not compile

KurtE, 2017-04-06:
The library is up at:
https://github.com/adafruit/Adafruit_SSD1306

I tried compiling their example for 128x64 spi and first off it did not want to find Wire library...
I imported the library again, and then,  ran into compile errors.
```
C:/Users/kurte/Documents/Arduino/libraries/Adafruit_SSD1306/examples/ssd1306_128x64_spi/ssd1306_128x64_spi.ino: In function 'void setup()':
C:/Users/kurte/Documents/Arduino/libraries/Adafruit_SSD1306/examples/ssd1306_128x64_spi/ssd1306_128x64_spi.ino:97:16: error: 'testdrawline' was not declared in this scope
   testdrawline();
                ^
C:/Users/kurte/Documents/Arduino/libraries/Adafruit_SSD1306/examples/ssd1306_128x64_spi/ssd1306_128x64_spi.ino:103:16: error: 'testdrawrect' was not declared in this scope
   testdrawrect();
                ^
C:/Users/kurte/Documents/Arduino/libraries/Adafruit_SSD1306/examples/ssd1306_128x64_spi/ssd1306_128x64_spi.ino:109:16: error: 'testfillrect' was not declared in this scope
   testfillrect();
                ^
C:/Users/kurte/Documents/Arduino/libraries/Adafruit_SSD1306/examples/ssd1306_128x64_spi/ssd1306_128x64_spi.ino:115:18: error: 'testdrawcircle' was not declared in this scope
   testdrawcircle();
                  ^
C:/Users/kurte/Documents/Arduino/libraries/Adafruit_SSD1306/examples/ssd1306_128x64_spi/ssd1306_128x64_spi.ino:126:21: error: 'testdrawroundrect' was not declared in this scope
   testdrawroundrect();
                     ^
C:/Users/kurte/Documents/Arduino/libraries/Adafruit_SSD1306/examples/ssd1306_128x64_spi/ssd1306_128x64_spi.ino:130:21: error: 'testfillroundrect' was not declared in this scope
   testfillroundrect();
                     ^
C:/Users/kurte/Documents/Arduino/libraries/Adafruit_SSD1306/examples/ssd1306_128x64_spi/ssd1306_128x64_spi.ino:134:20: error: 'testdrawtriangle' was not declared in this scope
   testdrawtriangle();
                    ^
C:/Users/kurte/Documents/Arduino/libraries/Adafruit_SSD1306/examples/ssd1306_128x64_spi/ssd1306_128x64_spi.ino:138:20: error: 'testfilltriangle' was not declared in this scope
   testfilltriangle();
                    ^
C:/Users/kurte/Documents/Arduino/libraries/Adafruit_SSD1306/examples/ssd1306_128x64_spi/ssd1306_128x64_spi.ino:143:16: error: 'testdrawchar' was not declared in this scope
   testdrawchar();
                ^
C:/Users/kurte/Documents/Arduino/libraries/Adafruit_SSD1306/examples/ssd1306_128x64_spi/ssd1306_128x64_spi.ino:149:18: error: 'testscrolltext' was not declared in this scope
   testscrolltext();
                  ^
C:/Users/kurte/Documents/Arduino/libraries/Adafruit_SSD1306/examples/ssd1306_128x64_spi/ssd1306_128x64_spi.ino:179:72: error: 'testdrawbitmap' was not declared in this scope
   testdrawbitmap(logo16_glcd_bmp, LOGO16_GLCD_HEIGHT, LOGO16_GLCD_WIDTH);
                                                                        ^
[Build] Error occurred.
```

Note: I have edited libraries, but program does build file on Arduino 1.8.2 Teensyduino latest, and T3.6 selected. 

It is like it is not autogenerating the appropriate functions forward references.

Robot-Will, 2017-04-07:
Fixed, still caused by macro, thanks,

closed 2017-04-08 by Robot-Will

buzzy, 2018-11-11:
I have the exact same problem now with PlatformIO with latest master. Any idea how to fix?

buzzy, 2018-11-11:
Ok, I found the problem. The example files are wrong. When I moved the setup() and loop() functions to the end of the file, it worked. You might want to update the examples.

KurtE, 2018-11-11:
@buzzy 
Just an FYI - I currently no longer use Stino as I kept running into memory issues with it and Stino did not follow the same rules as Arduino on which version of library was to be used...  

And that @Robot-Will has not updated the code, or issues in over a year and a half.

wavesailor, 2019-12-27:
> Ok, I found the problem. The example files are wrong. When I moved the setup() and loop() functions to the end of the file, it worked. You might want to update the examples.

Thanks ... That solved it for me