Robot-Will/Stino #418: At [Step 2] I get **[Error] No main source file found.**

JasonCDN, 2017-04-07:
When I attempt to build with Stino, it gives me an error. Here is the output:

```
[Build] C:/Users/CDN/Documents/_Code/2017/04...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Error] No main source file found. Main source file should be a c/c++ file, which contains main() function, or a Arduino file, which contains setup() and loop() functions.
```

When I Ctrl + `, I don't see any error messages at all. The last message is from package control saying "upgrade skipped". No additional messages appear when I try to build.

I am using the default Stino from package control (just installed today - 04/07/2017). My sublime text is Build 3126. I am operating Windows 10 Home Edition. I do have the Arduino IDE installed, but I am using the avr 1.6.18 toolchain (Board: Arduino Genuino/Uno).

Hopefully someone can help me out here 😄

Robot-Will, 2017-04-07:
Please show me the file "sketch.ino". Thanks.

JasonCDN, 2017-04-07:
Here it is. I'm using a library I wrote recently (DriveControl). It compiles normally under the Arduino IDE. Surely that can't be the problem?

```
#include <DriveControl.h>

DriveControl driver(65, 1);

void setup() {
	driver.addInstruction(21, 21);
};

void loop() {
	driver.run();
};
```

Note: I am able to build the blink sketch with Stino.

Robot-Will, 2017-04-07:
Remove ';' after '}', thanks.

```
#include <DriveControl.h>

DriveControl driver(65, 1);

void setup() {
	driver.addInstruction(21, 21);
}

void loop() {
	driver.run();
}
```

closed 2017-04-07 by Robot-Will