Robot-Will/Stino #419: Problem building avr c program with .h and .c files.

noumenonquest, 2017-04-07:
Problem building avr c program with a header file and a separate source code file(.c) to define the functions that are in the header file.
`[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[25.0%] Compiling main.c...
[50.0%] Compiling USART.c...
[75.0%] Linking everything together...
c:\users\user\appdata\local\arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.3-arduino2\bin\../lib/gcc/avr/4.9.2/../../../../avr/bin/ar.exe: C:/Users/user/AppData/Local/Arduino15/build/example/core/core.a: No such file or directory
[Build] Error occurred.
`

Robot-Will, 2017-04-07:
I need the files to test, I have no experience about avr projects. Thanks.

Robot-Will, 2017-04-07:
I fixed. Thanks.

```
[Build] D:/Documents/Arduino/avr01...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[25.0%] Compiling avr01.ino.cpp...
[50.0%] Compiling USART.c...
[75.0%] Linking everything together...
[100.0%] Creating binary files...
Sketch uses 292 bytes (0.1%) of program storage space. Maximum is 253,952 bytes.
Global variables use 0 bytes (0.0%) of dynamic memory, leaving 8,192 bytes for local variables. Maximum is 8,192 bytes.
Build done.
```

noumenonquest, 2017-04-07:
now i get the following error.

> [Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[20.0%] Compiling ...
avr-g++: error: : No such file or directory
avr-g++: fatal error: no input files
compilation terminated.
[Build] Error occurred.

Robot-Will, 2017-04-08:
Thanks for your sketch, I fixed.

```
[Build] D:/Documents/Arduino/main...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[25.0%] Compiling main.c...
[50.0%] Compiling USART.c...
[75.0%] Linking everything together...
[100.0%] Creating binary files...
Sketch uses 266 bytes (0.8%) of program storage space. Maximum is 32,256 bytes.
Global variables use 16 bytes (0.8%) of dynamic memory, leaving 2,032 bytes for local variables. Maximum is 2,048 bytes.
Build done.

```

closed 2017-04-08 by noumenonquest