Robot-Will/Stino #420: An #include doesn't work: No such file or directory

Titanium-TechFR, 2017-04-08:
Hello,
I'm using Sublime Text 3 (build 3126) on Windows 10 (64 bit) and I have an issue when I want to include a library.

In my ino file:
`#include <LiquidCrystal.h>`

I have:
```
[Build] D:/Users/antho/SVN/aSVN/LifeCompagnion...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[3.4%] Compiling LifeCompagnion.ino.cpp...
D:/Users/antho/SVN/aSVNE/LifeCompagnion/LifeCompagnion.ino:120:27: fatal error: LiquidCrystal.h: No such file or directory
 #include <LiquidCrystal.h>
                           ^
compilation terminated.
[Build] Error occurred.
```

I have tested to put the library. In my document folder (Documents\Arduino\libraries). I have try to install the library with Arduino > Install library, but, it doesn't work.

Sorry if my english is not good.

Titanium TechFR

Robot-Will, 2017-04-09:
Stino-Dev branch, ST 3126, Win10x64, Arduino Uno, Blink example

LiquidCrystal lib is in D:\Documents\Arduino\libraries\LiquidCrystal

I passed build without error. 

```
#include <LiquidCrystal.h>

int led_pin = 13;
void setup()
{
    // initialize digital pin LED_BUILTIN as an output.
    pinMode(led_pin, OUTPUT);
    Serial.begin(9600);
}

// the loop function runs over and over again forever
void loop()
{
    Serial.println("Hello World!");
    digitalWrite(led_pin, HIGH); // turn the LED on (HIGH is the voltage level)
    delay(1000); // wait for a second
    digitalWrite(led_pin, LOW); // turn the LED off by making the voltage LOW
    delay(1000); // wait for a second
}
```

```
[Build] D:/Documents/Arduino/Blink...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[3.4%] Compiling Blink.ino.cpp...
[6.9%] Compiling LiquidCrystal.cpp...
[10.3%] Compiling abi.cpp...
[13.8%] Compiling CDC.cpp...
[17.2%] Compiling HardwareSerial.cpp...
[20.7%] Compiling HardwareSerial0.cpp...
[24.1%] Compiling HardwareSerial1.cpp...
[27.6%] Compiling HardwareSerial2.cpp...
[31.0%] Compiling HardwareSerial3.cpp...
[34.5%] Compiling IPAddress.cpp...
[37.9%] Compiling main.cpp...
[41.4%] Compiling new.cpp...
[44.8%] Compiling PluggableUSB.cpp...
[48.3%] Compiling Print.cpp...
[51.7%] Compiling Stream.cpp...
[55.2%] Compiling Tone.cpp...
[58.6%] Compiling USBCore.cpp...
[62.1%] Compiling WMath.cpp...
[65.5%] Compiling WString.cpp...
[69.0%] Compiling hooks.c...
[72.4%] Compiling WInterrupts.c...
[75.9%] Compiling wiring.c...
[79.3%] Compiling wiring_analog.c...
[82.8%] Compiling wiring_digital.c...
[86.2%] Compiling wiring_pulse.c...
[89.7%] Compiling wiring_shift.c...
[93.1%] Compiling wiring_pulse.S...
[96.6%] Linking everything together...
[100.0%] Creating binary files...
Sketch uses 1,934 bytes (6.0%) of program storage space. Maximum is 32,256 bytes.
Global variables use 198 bytes (9.7%) of dynamic memory, leaving 1,850 bytes for local variables. Maximum is 2,048 bytes.
Build done.
```

Or you show me your sketch, and I will find out the reason. Thanks.

Titanium-TechFR, 2017-04-09:
Hello!
Thank you for your answer!
I fixed the error. The problem was that `#include <SwitchEncoder.h>` needs `#include <HowToSetup.h>` too. The Arduino IDE accept `#include <SwitchEncoder.h>` alone. When I'm using Arduino > Import library > SwitchEncoder-master, your package add both `#include`.

Thank you very much!
Your package is awesome!

Titanium TechFR

closed 2017-04-09 by Titanium-TechFR