Robot-Will/Stino #423: #defines used in memory allocation causes compile error...

KurtE, 2017-04-09:
Short version: 
This compiles without error
```
uint8_t screenmemory[512] = {0};
```
However this one errors out saying that screenmemory was already defined (at the same line number)
```
#define LCDWIDTH 128
#define LCDHEIGHT 32
uint8_t screenmemory[(LCDHEIGHT * LCDWIDTH) /8] = {0};
```

I can paste in whole program if you would like

Robot-Will, 2017-04-10:
I added the 3 lines to the blink sample and found no error. Or you show me the whole program. Thanks.

```
#define LCDWIDTH 128
#define LCDHEIGHT 32
uint8_t screenmemory[(LCDHEIGHT * LCDWIDTH) /8] = {0};

int led_pin = 13;
void setup()
{
    // initialize digital pin LED_BUILTIN as an output.
    pinMode(led_pin, OUTPUT);
    Serial.begin(9600);
}

// the loop function runs over and over again forever
void loop()
{
    Serial.println("Hello World!");
    digitalWrite(led_pin, HIGH); // turn the LED on (HIGH is the voltage level)
    delay(1000); // wait for a second
    digitalWrite(led_pin, LOW); // turn the LED off by making the voltage LOW
    delay(1000); // wait for a second
}
```

```
[Build] D:/Documents/Arduino/Blink...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[1.8%] Compiling Blink.ino.cpp...
[3.6%] Compiling AudioStream.cpp...
[5.5%] Compiling avr_emulation.cpp...
[7.3%] Compiling DMAChannel.cpp...
[9.1%] Compiling HardwareSerial1.cpp...
[10.9%] Compiling HardwareSerial2.cpp...
[12.7%] Compiling HardwareSerial3.cpp...
[14.5%] Compiling HardwareSerial4.cpp...
[16.4%] Compiling HardwareSerial5.cpp...
[18.2%] Compiling HardwareSerial6.cpp...
[20.0%] Compiling IntervalTimer.cpp...
[21.8%] Compiling IPAddress.cpp...
[23.6%] Compiling main.cpp...
[25.5%] Compiling new.cpp...
[27.3%] Compiling Print.cpp...
[29.1%] Compiling Stream.cpp...
[30.9%] Compiling Tone.cpp...
[32.7%] Compiling usb_audio.cpp...
[34.5%] Compiling usb_flightsim.cpp...
[36.4%] Compiling usb_inst.cpp...
[38.2%] Compiling WMath.cpp...
[40.0%] Compiling WString.cpp...
[41.8%] Compiling yield.cpp...
[43.6%] Compiling analog.c...
[45.5%] Compiling eeprom.c...
[47.3%] Compiling keylayouts.c...
[49.1%] Compiling math_helper.c...
[50.9%] Compiling mk20dx128.c...
[52.7%] Compiling nonstd.c...
[54.5%] Compiling pins_teensy.c...
[56.4%] Compiling serial1.c...
[58.2%] Compiling serial2.c...
[60.0%] Compiling serial3.c...
[61.8%] Compiling serial4.c...
[63.6%] Compiling serial5.c...
[65.5%] Compiling serial6.c...
[67.3%] Compiling serial6_lpuart.c...
[69.1%] Compiling ser_print.c...
[70.9%] Compiling touch.c...
[72.7%] Compiling usb_desc.c...
[74.5%] Compiling usb_dev.c...
[76.4%] Compiling usb_joystick.c...
[78.2%] Compiling usb_keyboard.c...
[80.0%] Compiling usb_mem.c...
[81.8%] Compiling usb_midi.c...
[83.6%] Compiling usb_mouse.c...
[85.5%] Compiling usb_mtp.c...
[87.3%] Compiling usb_rawhid.c...
[89.1%] Compiling usb_seremu.c...
[90.9%] Compiling usb_serial.c...
[92.7%] Compiling usb_touch.c...
[94.5%] Compiling memcpy-armv7m.S...
[96.4%] Compiling memset.S...
[98.2%] Linking everything together...
[100.0%] Creating binary files...
Opening Teensy Loader...
Sketch uses 24,628 bytes (2.3%) of program storage space. Maximum is 1,048,576 bytes.
Global variables use 5,040 bytes (1.9%) of dynamic memory, leaving 257,104 bytes for local variables. Maximum is 262,144 bytes.
[Build done.]
```

KurtE, 2017-04-10:
Here is the whole program.  

```

// Quick and dirty extract of parts of Teensyview to compare speed of different SPI methods to compare

// speed.

#include <SPI.h>

// #include <SPIKinetis.h>

#define LCDWIDTH 128

#define LCDHEIGHT 32

#define PIN_RESET 15

#define PIN_SCK 13

#define PIN_MOSI 11

#define PIN_DC 9

#define PIN_CS 10

 

uint8_t pcs_data = 0;

uint8_t pcs_command = 0;

 

//uint8_t screenmemory[512] = {0};

 

uint8_t screenmemory[(LCDHEIGHT * LCDWIDTH) /8] = {0};

uint8_t _height = 32;

volatile uint8_t * _csport, * _dcport;

uint8_t _cspinmask, _dcpinmask;

uint32_t clockRateSetting = 8000000;

void setup()

{

                pinMode(PIN_CS, OUTPUT);

                _csport = portOutputRegister(digitalPinToPort(PIN_CS));

                _cspinmask = digitalPinToBitMask(PIN_CS);

                * _csport |= _cspinmask;

                pinMode(PIN_DC, OUTPUT);

                _dcport = portOutputRegister(digitalPinToPort(PIN_DC));

                _dcpinmask = digitalPinToBitMask(PIN_DC);

                SPI.begin();

                for (uint16_t i = 0; i < sizeof(screenmemory); i++)

[... 418 lines cut ...]
                                w = (*pscreen_data++) << 8;

                                                w |= *pscreen_data++;

                                                if (pscreen_data >= &screenmemory[(LCDWIDTH*LCDHEIGHT)/8])

                                                                writedata16_last(w);

                                                else

                                                                writedata16_cont(w);

                                }

                }

                * _csport |= _cspinmask;

                SPI.endTransaction();

}

#endif

```

Robot-Will, 2017-04-11:
I am not sure whether the problem is solved because I got the same build errors from both this plugin and Arduino IDE. I need your feedback. Thanks.

```
D:/Documents/Arduino/test00/test00.ino: In function 'void display_transfer_buf()':
D:/Documents/Arduino/test00/test00.ino:84:84: error: no matching function for call to 'SPIClass::transfer(uint8_t [4], NULL, unsigned int)'
         SPI.transfer(set_page_column_address, NULL, sizeof(set_page_column_address));
                                                                                    ^
In file included from D:/Documents/Arduino/test00/test00.ino:3:0:
E:/dev/arduino/arduino-1.8.2/hardware/teensy/avr/libraries/SPI/SPI.h:479:24: note: candidate: static uint8_t SPIClass::transfer(uint8_t)
  inline static uint8_t transfer(uint8_t data) {
                        ^
E:/dev/arduino/arduino-1.8.2/hardware/teensy/avr/libraries/SPI/SPI.h:479:24: note:   candidate expects 1 argument, 3 provided
E:/dev/arduino/arduino-1.8.2/hardware/teensy/avr/libraries/SPI/SPI.h:491:14: note: candidate: static void SPIClass::transfer(void*, size_t)
  static void transfer(void *buf, size_t count);
              ^
E:/dev/arduino/arduino-1.8.2/hardware/teensy/avr/libraries/SPI/SPI.h:491:14: note:   candidate expects 2 arguments, 3 provided
D:/Documents/Arduino/test00/test00.ino:87:50: error: no matching function for call to 'SPIClass::transfer(uint8_t*&, NULL, int)'
         SPI.transfer(pscreen_data, NULL, LCDWIDTH);
                                                  ^
In file included from D:/Documents/Arduino/test00/test00.ino:3:0:
E:/dev/arduino/arduino-1.8.2/hardware/teensy/avr/libraries/SPI/SPI.h:479:24: note: candidate: static uint8_t SPIClass::transfer(uint8_t)
  inline static uint8_t transfer(uint8_t data) {
                        ^
E:/dev/arduino/arduino-1.8.2/hardware/teensy/avr/libraries/SPI/SPI.h:479:24: note:   candidate expects 1 argument, 3 provided
E:/dev/arduino/arduino-1.8.2/hardware/teensy/avr/libraries/SPI/SPI.h:491:14: note: candidate: static void SPIClass::transfer(void*, size_t)
  static void transfer(void *buf, size_t count);
              ^
E:/dev/arduino/arduino-1.8.2/hardware/teensy/avr/libraries/SPI/SPI.h:491:14: note:   candidate expects 2 arguments, 3 provided
```

KurtE, 2017-04-11:
Thanks, I am working on an updated SPI library.  Hopefully trying to improve the performance… Added new api, which you did not have….

 

I synced up this morning, tried compiling with the macros and it appears to work now 😃 

 

Kurt

closed 2017-04-12 by Robot-Will