Robot-Will/Stino #425: STM32 won't complete build

Defragster, 2017-04-10:
Stuck at: "[Step 2] Find all source files."

reloading plugin Stino-Stino2017.StinoCommands
plugins loaded
Package Control: Skipping automatic upgrade, last run at 2017-04-09 22:11:34, next run at 2017-04-09 23:11:34 or after
Exception in thread Thread-14:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "C:\Users\Tim\AppData\Roaming\Sublime Text 3\Packages\Stino-Stino2017\libs\base_utils\task_queue.py", line 86, in _run
    self._consumer(*args)
  File "C:\Users\Tim\AppData\Roaming\Sublime Text 3\Packages\Stino-Stino2017\libs\stino_runtime\__init__.py", line 2159, in build_sketch
    main_file_path = prj.get_combine_path()
  File "C:\Users\Tim\AppData\Roaming\Sublime Text 3\Packages\Stino-Stino2017\libs\base_utils\c_project.py", line 320, in get_combine_path
    minus_src_path, self._is_arduino_project)
  File "C:\Users\Tim\AppData\Roaming\Sublime Text 3\Packages\Stino-Stino2017\libs\base_utils\c_project.py", line 129, in combine_ino_files
    if os.path.isfile(minus_src_path):
  File "./python3.3/genericpath.py", line 29, in isfile
TypeError: can't specify None for path argument

Defragster, 2017-04-10:
Opened IDE and confirmed sketch builds and uploads

Robot-Will, 2017-04-10:
Please try the lateast version of dev branch. If still the same problem, leave message here. Thanks.

In line 104, before the default value of minus_src_path is `None`, and now it is `''`. 
```
def combine_ino_files(ino_file_paths, target_file_path,
                      minus_src_path='', is_arduino_project=True):
```

And I have no error for ladybug borad.

```
[Build] D:/Documents/Arduino/Blink...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[3.3%] Compiling Blink.ino.cpp...
[6.7%] Compiling abi.cpp...
[10.0%] Compiling CDC.cpp...
[13.3%] Compiling FS.cpp...
[16.7%] Compiling IPAddress.cpp...
[20.0%] Compiling main.cpp...
[23.3%] Compiling new.cpp...
[26.7%] Compiling Print.cpp...
[30.0%] Compiling RingBuffer.cpp...
[33.3%] Compiling STM32.cpp...
[36.7%] Compiling Stream.cpp...
[40.0%] Compiling Uart.cpp...
[43.3%] Compiling USBCore.cpp...
[46.7%] Compiling WMath.cpp...
[50.0%] Compiling WString.cpp...
[53.3%] Compiling dtostrf.c...
[56.7%] Compiling eeprom.c...
[60.0%] Compiling fdevopen.c...
[63.3%] Compiling hooks.c...
[66.7%] Compiling itoa.c...
[70.0%] Compiling stm32l4_wiring.c...
[73.3%] Compiling stm32l4_wiring_analog.c...
[76.7%] Compiling stm32l4_wiring_digital.c...
[80.0%] Compiling stm32l4_wiring_interrupts.c...
[83.3%] Compiling stm32l4_wiring_pulse.c...
[86.7%] Compiling stm32l4_wiring_shift.c...
[90.0%] Compiling stm32l4_wiring_tone.c...
[93.3%] Compiling variant.cpp...
[96.7%] Linking everything together...
[100.0%] Creating binary files...
Sketch uses 22,136 bytes (9.0%) of program storage space. Maximum is 245,760 bytes.
[Build done.]
```

Defragster, 2017-04-10:
I updated to latest of DEV branch before posting.  I haven't compiled in a week or so for this board.

I wiped my Sublime Text 3\Packages\Stino... folder and copied in from github folder.

Defragster, 2017-04-10:
Using :: [Build] I:/arduino-1.8.1/examples/01.Basics/Blink...
[Step 1] Check Toolchain.
[Step 2] Find all source files.

I see the same:
Package Control: No updated packages
Exception in thread Thread-16:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "C:\Users\xxx\AppData\Roaming\Sublime Text 3\Packages\Stino-Stino2017\libs\base_utils\task_queue.py", line 86, in _run
    self._consumer(*args)
  File "C:\Users\xxx\AppData\Roaming\Sublime Text 3\Packages\Stino-Stino2017\libs\stino_runtime\__init__.py", line 2159, in build_sketch
    main_file_path = prj.get_combine_path()
  File "C:\Users\xxx\AppData\Roaming\Sublime Text 3\Packages\Stino-Stino2017\libs\base_utils\c_project.py", line 320, in get_combine_path
    minus_src_path, self._is_arduino_project)
  File "C:\Users\xxx\AppData\Roaming\Sublime Text 3\Packages\Stino-Stino2017\libs\base_utils\c_project.py", line 129, in combine_ino_files
    if os.path.isfile(minus_src_path):
  File "./python3.3/genericpath.py", line 29, in isfile
TypeError: can't specify None for path argument

Defragster, 2017-04-10:
in: C:\Users\xxx\AppData\Roaming\Sublime Text 3\Packages\Stino-Stino2017\libs\base_utils\c_project.py
{ this is not my github 'dev' tree - but when I installed stino it was then 'Packages\Stino-Stino2017' and I left it that way. }

I see:
def combine_ino_files(ino_file_paths, target_file_path,
                      minus_src_path='', is_arduino_project=True):

Robot-Will, 2017-04-10:
Sorry, I found another dedault value in line 312

`def get_combine_path(self, minus_src_path=''):`

I fixed, and you try it, thanks.

Defragster, 2017-04-10:
That FIXED it.
Good Work!

closed 2017-04-10 by Defragster

kourindouhime, 2018-10-24:
No that doesn't fix it