Robot-Will/Stino #428: compile error

acs8369, 2017-04-11:
I got an error when try to compile default example from SPI libraries

[Build] C:/Users/xxx/AppData/Local/Arduino15/packages/arduino/hardware/avr/1.6.17/libraries/SPI/examples/BarometricPressureSensor...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[3.4%] Compiling BarometricPressureSensor.ino.cpp...
C:/Users/Nattawut/AppData/Local/Arduino15/packages/arduino/hardware/avr/1.6.17/libraries/SPI/examples/BarometricPressureSensor/BarometricPressureSensor.ino: In function 'void setup()':
C:/Users/Nattawut/AppData/Local/Arduino15/packages/arduino/hardware/avr/1.6.17/libraries/SPI/examples/BarometricPressureSensor/BarometricPressureSensor.ino:51:27: error: 'writeRegister' was not declared in this scope
   writeRegister(0x02, 0x2D);
                           ^
C:/Users/Nattawut/AppData/Local/Arduino15/packages/arduino/hardware/avr/1.6.17/libraries/SPI/examples/BarometricPressureSensor/BarometricPressureSensor.ino: In function 'void loop()':
C:/Users/Nattawut/AppData/Local/Arduino15/packages/arduino/hardware/avr/1.6.17/libraries/SPI/examples/BarometricPressureSensor/BarometricPressureSensor.ino:60:27: error: 'writeRegister' was not declared in this scope
   writeRegister(0x03, 0x0A);
                           ^
C:/Users/Nattawut/AppData/Local/Arduino15/packages/arduino/hardware/avr/1.6.17/libraries/SPI/examples/BarometricPressureSensor/BarometricPressureSensor.ino:65:40: error: 'readRegister' was not declared in this scope
     int tempData = readRegister(0x21, 2);
                                        ^
[Build] Error occurred.

------------------------------------------------------------

[Build] D:/Software/arduino-1.8.2/hardware/arduino/avr/libraries/SPI/examples/BarometricPressureSensor...
[Step 1] Check Toolchain.
[Step 2] Find all source files.
[Step 3] Start building.
[3.4%] Compiling BarometricPressureSensor.ino.cpp...
D:/Software/arduino-1.8.2/hardware/arduino/avr/libraries/SPI/examples/BarometricPressureSensor/BarometricPressureSensor.ino: In function 'void setup()':
D:/Software/arduino-1.8.2/hardware/arduino/avr/libraries/SPI/examples/BarometricPressureSensor/BarometricPressureSensor.ino:51:27: error: 'writeRegister' was not declared in this scope
   writeRegister(0x02, 0x2D);
                           ^
D:/Software/arduino-1.8.2/hardware/arduino/avr/libraries/SPI/examples/BarometricPressureSensor/BarometricPressureSensor.ino: In function 'void loop()':
D:/Software/arduino-1.8.2/hardware/arduino/avr/libraries/SPI/examples/BarometricPressureSensor/BarometricPressureSensor.ino:60:27: error: 'writeRegister' was not declared in this scope
   writeRegister(0x03, 0x0A);
                           ^
D:/Software/arduino-1.8.2/hardware/arduino/avr/libraries/SPI/examples/BarometricPressureSensor/BarometricPressureSensor.ino:65:40: error: 'readRegister' was not declared in this scope
     int tempData = readRegister(0x21, 2);
                                        ^
[Build] Error occurred.

--------------------------------------------------------------

Robot-Will, 2017-04-12:
Fixed. Thanks.

acs8369, 2017-04-12:
thank you, stino help me write 1000+ line of code with ease.

closed 2017-04-12 by Robot-Will