Robot-Will/Stino #431: Open Sketch in new folder: multi file sketch

KurtE, 2017-04-12:
It is probably by design, but thought I would mention. 

I decided to try out the feature to open sketch in new window, which is great.  However in this case the sketch I tried out (https://github.com/KurtE/WellMonitor) has maybe 9 source files, with one .ino file and the others are .h or .cpp files.

When the new window opened up, It only showed the .ino file.  And the sidebar appears to be disabled in this new window, so not sure how best to open up any of the other files.   Obviously I could probably try the file Open menu.  The only other way is to start a compile and if there are any errors or warnings, those files appear to show up as tabs in the new window. 

Again maybe by design.

Robot-Will, 2017-04-13:
I added a menu item:

Menu -> Open Sketch -> All Source Files

Try it, thanks.

KurtE, 2017-04-13:
That helps, especially if you are proactive and turn it on before you open your sketch.

closed 2017-04-14 by Robot-Will